Ultimaker/Cura #6265: Ender 3 settings and time estimation wrong in Cura 4.2

Vashnevskaya, 2019-08-27:
**Application version**
Ultimaker Cura 4.2

**Platform**
Windows 10

**Printer**
Ender 3

**Reproduction steps**
Default settings seem to be wrong. The estimated time for the same print rised from 1'45" to 2'30", as well as increasing the weight of the piece.

**Actual results**
With Cura 4.0 and 4.1 the print came out with the same weight, 8gr, and the same real time, 1'30", although Cura 4.0 estimated 1'20" and Cura 4.1 estimated 1'45". But with Cura 4.2, the very same piece came out weighting more, and actually took 2'30"

**Expected results**
Real time and weight should have been the same.

**Additional information**
The default settings seem different also. The bed temperature is set by default to 50º instead of 60, which has proven to be low enough for the print to always detach from the bed, even using lacquer. Never happened at 60º, by the way.

rburema, 2019-08-28:
Hello @Vashnevskaya ,

 - Settings for Creality-printers have changed since we pulled in the creawsome-mod. 
 - Estimates can be off, that's why they're estimates ... _however_ this is a bit of an extreme case, so it might be worth looking into. Just to confirm: You said the estimates in 4.0 and 4.1 where lower, was the estimate in 4.2 also +/- 1:45? Could you attach the project file?

Vashnevskaya, 2019-08-28:
Hi, and thanks for the answer!

  The settings were just the default settings for either 4.0, 4.1, and 4.2.
The only change I made was choosing a 0'16mm layer height, and 20% infill
with grid shape. Aside that, the estimations were made with the default
program settings.

  Estimations:
Cura 4.0 - 8gr - 1'20"
Cura 4.1 - 8gr - 1'49"
Cura 4.2 - 11gr - 2'48"

  Results:
Cura 4.0 - 1'30"
Cura 4.1 - 1'30"
Cura 4.0 - 2'25", and heavier resulting piece

I've tried to attach the file I used, but your mail says it's too big.
The archive is "TMGenericCityMaster2019.stl", taken from here:
https://www.thingiverse.com/thing:3153866/files

Kind regards
Carlos

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Libre
de virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Vashnevskaya, 2019-08-28:
> (I assume you mean '4.2' in the last item of the result column.)
>
> It seems that the estimation (though a bit off) is not really the issue
> here, since it show a longer time and a heavier model to begin with.
>
> The change in time is likely because we radically changed all the profiles
> for Creality printers, since we pulled in the 'Creawsome' mod. That means
> some default settings (for those printers) will have changed, which causes
> the increased print-time (and weight).

Yep, I meant 4.2 for the last one.

Does it mean you'll review the settings, so it'll have similar parameters
as in previous versions, or you'll mantain the new ones?

In the latter case, would it be possible to change printing settings, so
the result would be the same as in a Cura 4.1 slicing?

Kind regards

rburema, 2019-08-28:
(I assume you mean '4.2' in the last item of the result column.)

It seems that the estimation (though a bit off) is not really the issue here, since it show a longer time and a heavier model to begin with.

The change in time is likely because we radically changed all the profiles for Creality printers, since we pulled in the 'Creawsome' mod. That means some default settings (for those printers) will have changed, which causes the increased print-time (and weight).

bbrowaros, 2019-08-28:
Hi. I've notice similar problem however for me the estimation is about 50% of estimation from octoprint. With cura 4.1 I had a 25 % max difference however with a update to 4.2  cura is showing me 3h 20min however once the file is loaded to octopront it shows 6h 40 min.

Liger0, 2019-08-30:
Can you do a test forcing in cura to use cura's jerk and acceleration?
You probably have inappropriate values in the firmware.

Ghostkeeper, 2019-09-02:
If the time estimates of Cura match the actual printing time, there is nothing that Cura can improve in this regard. Your test indicates a difference of 2h48 vs. 2h25 which is about 14%. This would be too much for Ultimaker's printers since we assume a 5% fluctuation in Marlin's stepper timing. However for your Ender 3 I have no idea what fluctuations you get. The difference could be explained by Cura's defaults not matching up with your firmware's defaults for acceleration, jerk, maximum/minimum feedrates, etc. This can be amended by enabling Jerk and Acceleration control and further specifying the machine settings for max feedrate to match your firmware.

However the problem that Vashnevskaya seems to want to indicate is with the actual settings printing more towards quality than productivity than before. These profiles are not maintained by Ultimaker though but by the community of Creality printer users. If you have any suggestions to improve them, I think you should post them here: https://github.com/Ultimaker/Cura/issues/6106

I'll close this ticket since half of it is expected behaviour and the other half duplicates #6106.

closed 2019-09-02 by Ghostkeeper

oofiksoo, 2019-09-07:
This was affected by the Creawsome Mod. Previous version of the software reported estimations within 5-11 minutes very reliably. Now it is off by 30 minutes or more. I see this issue is already closed, and the response being, We don't care about your printer, or your experience. This is ridiculous. I stand by my adversity to Cura. I was really hoping this was something you were actually willing to address based on your previous posts in response to my concerns. I can see I was on a fools errand to alter my perspective. It is blatantly obvious that the Ultimaker position is to kill the competition by making their users printing experience miserable, if they even get that far.

nallath, 2019-09-09:
No, the issue was closed because it's a duplicate and a notification that the settings don't match up with whatever is in the firmware.

If anyone tells us what the values should be, we'd be more than happy to add them. 

As a final note; We're not here to do the job of Creality. They choose to not develop anything with or for Cura. It's not our responsibility to ensure that their printers work with Cura or to provide support for their printers. I'd welcome their contributions, as I would with any contribution. But so far they have completely absent. So if you want those changes, it's Creality you need to be complaining at, not us (or do it yourself/ convince someone in the community to make the pull requests).