Ultimaker/Cura #6268: Problem with select model

userosos, 2019-08-27:
Cura 4.2.1 have problem with select model. Sometimes i can not select a model into bed. And can't rotate selected model.

rburema, 2019-08-28:
Hello @userosos, Does this happen for specific models, or all of them?

userosos, 2019-08-28:
@rburema all of them

rburema, 2019-08-28:
> Sometimes i can not select a model into bed.

Do you mean that the model is below the build-plate when you try to select it?

userosos, 2019-08-28:
@rburema 
> Do you mean that the model is below the build-plate when you try to select it?
Yes. I has the problem regularly but not all of them models which are located on the build-plate.

rburema, 2019-08-29:
So if I understand correctly, the following .gif illustrates the problem?

[image]

The gray area is the 'shadow' of the model, not the actual model, so it can't be used to select it. Which is more difficult, since the object is not visible below the build-plate?

userosos, 2019-08-29:
> 
> 
> So if I understand correctly, the following .gif illustrates the problem?
> 
> [image]
> 
> The gray area is the 'shadow' of the model, not the actual model, so it can't be used to select it. Which is more difficult, since the object is not visible below the build-plate?

No. I can not select model sometimes. But i can not replay the bug now.

fieldOfView, 2019-08-29:
Are you using the Sidebar GUI plugin?

userosos, 2019-08-30:
> Are you using the Sidebar GUI plugin?

Yes. I use the plugin.

fieldOfView, 2019-08-30:
I can sometimes reproduce it.
I'm looking into it.

Ghostkeeper, 2019-09-03:
I've had this problem for the past 2 years. There seem to be some invisible zones in the 3D scene where the selection area wasn't properly rendered or something.

fieldOfView, 2019-09-03:
I have sofar been unable to reproduce it with https://github.com/Ultimaker/Uranium/pull/518 in place. 

I am not sure if the selectionpass is to blame; I inspected the selectionpass with RenderDoc (an OpenGL debugger) and the selectionpass rendered fine. It might be that something that is not currently visible is interfering with the mouse coordinates.

Ghostkeeper, 2019-09-03:
@rburema has had a similar issue on a branch that he was working on and that was fixed by Ultimaker/Uranium#518 as well. That gives me sufficient confidence that this issue is also fixed by it. Thanks fieldOfView!

closed 2019-09-03 by Ghostkeeper