Ultimaker/Cura #6271: [4.2.1] Model doesn't touch the base in the view and with a shadow mid air

Liger0, 2019-08-28:
**Application version**
4.2.1

**Platform**
Win10x64

**Printer**
Cr10s

**Reproduction steps**
Load my project

**Actual results**
The model and skirt shown in the air with a gap from the bed with the shadow mid air, even tho the coordinate is 0.

**Expected results**
Model touching the build plate and shadow on the bed.

**Additional information**
[CCR10S_pin.zip](https://github.com/Ultimaker/Cura/files/3550251/CCR10S_pin.zip)
[image]
[image]
[image]

rburema, 2019-09-03:
I can reproduce this on our end. Seems to happen to other models and printers as well, but doesn't get noticed unless you look at a very small model.

fieldOfView, 2019-09-03:
A small offset between the buildplate and the bottom of models is added to prevent z-fighting between the buildplate and the (convex hull) shadows.

rburema, 2019-09-03:
@Liger0 Sorry, we've decided to defer this for now.

Liger0, 2019-09-03:
> 
> 
> A small offset between the buildplate and the bottom of models is added to prevent z-fighting between the buildplate and the (convex hull) shadows.

I thougt so, but the shadow in on the middle of the Z of the model.

Ghostkeeper, 2019-09-05:
Indeed, it's 0.1mm above the build plate as per this line here:

https://github.com/Ultimaker/Cura/blob/e02f0f1480551b5131c94f4741ba36307c70d9eb/cura/Scene/ConvexHullNode.py#L40

This has already been tuned. If it's any lower you'll start getting Z fighting with common printer sizes.

A proper fix would include render tricks to ignore the Z coordinate when drawing the convex hull node, but we don't have that system in place now.

Liger0, 2019-09-05:
I udnerstand the model above the plate, but as I said, the shadow is way upper.

closed 2023-06-24 by github-actions[bot]