Ultimaker/Cura #6273: Cura 4.2.1  Configuration errors

Thisismydigitalself, 2019-08-28:
**Application version**
Cura 4.2.1

**Platform**
Win10/64/i5/16Gb/NV 750TI/

**Printer**
CFFFP

**Reproduction steps**
Reload *.3mf projects created with 4.2.1

**Actual results**
[image]

**Expected results**
I am here to ask a very simple question: When will be the last time I see these errors?
Why 3.6 was and apparently the one and only version which never prompt me these errors?

**Additional information**
I can't say I was happy to see these errors are back - i was hoping I would never see those again.

Thisismydigitalself, 2019-08-28:
Log file contains private information. 
if you want me to look for specific error in it let me know what to search for.

Thisismydigitalself, 2019-08-31:
Everyday the list grows bigger. I was hoping you guys can release a hot fix or at least tell us how to clear this error message, but not even one comment to this reported repeated issue.
[image]

LipuFei, 2019-09-02:
@Thisismydigitalself Could you provide the 3MF project file that can reproduce this issue? Thanks.

Thisismydigitalself, 2019-09-02:
i am not at my desk. could this be due to the fact that i jumped from 3.6 to 4.2.1?
i have no intention on installing any previous 4.x versions but if you think this is mandatory, i will do that only for fixing this issue

Thisismydigitalself, 2019-09-02:
@LipuFei 
3mf attached (change TEST.txt to TEST.3mf).
All i did was, within 4.2.1, saved a project and reloaded back. 
Please let me know what goes on cause the list of errors is now HUGE and i need a fix without resetting all my 4.2.1 profiles/settings.
[TEST.txt](https://github.com/Ultimaker/Cura/files/3566334/TEST.txt)

[image]

Thisismydigitalself, 2019-09-02:
Well, things are worst that I thought. now when the list of errors is so long, I can't use 4.2.1 anymore as there is no way to close it and see the print bed.

Wow. I guess I am pushed to go back to 3.6.
What a mess! i don't know how to use my current 4.2.1 profiles with 3.6.

This is a **BIG** issue. 
[image]

rburema, 2019-09-03:
I see a config error when reloading the saved project, but just a single one. This is indeed bad.

LipuFei, 2019-09-03:
I found out why. The ContainerRegistry is not performing correct searching for names with special characters such as `[]`. They were reserved for special purposes but not taken into account the general fields such as "name", which can be any user input.

See:
https://github.com/Ultimaker/Uranium/blob/07e99d7a1afbfd70837395fe270549e535dc2f1d/UM/Settings/ContainerQuery.py#L66-L86

Thisismydigitalself, 2019-09-03:
thx. going to try to see if i have access to edit profile names. is this new to 4.x? i use this format since 3.6 and cant recall ever having this issue

Thisismydigitalself, 2019-09-03:
@LipuFei 
Thank you, renamed all my profile's names, replacing '[ ]' with '-' and now Cura loads without errors. Now I need to reload all my 3mf projects which are based on old named profiles and see what will happen. I should hope this will work as if not - this means troubles.

Thisismydigitalself, 2019-09-03:
FYI, loaded archived 3mf with [ ] named profile. this time chose to create new printer and new material. project loaded OK, there is now a new profile with [ ] in its name, and when i relaunch Cura - no error. so now even when there is a [ ] names profile in the list of profiles - no errors when Cura loads. i am confused. before my next test - this time UPDATING printer and profile during import.
Before I do so, I will backup all my new profiles.

Thisismydigitalself, 2019-09-03:
This is now when i reload a 3mf project containing [ ] in it's associated profile. this time I chose to update printers and materials and - I got this error.
Reloading projects is and always was tricky. Cura is not dealing with projects as good as I would have hoped for.
[image]

LipuFei, 2019-09-03:
Fixed in 5b1bbb0764ce0e05c3582c2ac9884909db02845d. It wasn't too difficult.

LipuFei, 2019-09-03:
@Thisismydigitalself If you rename your quality profiles to something that doesn't start and end with `[]`, it should be ok. Another problem is that if you have seen this error message after loading a project file, it will create duplicate quality profiles, which causes the error message to occur. If you clean up your `cura` directory to have a clean start, and do not load any project file **more than once** which contains a `[*]` pattern custom quality file, you can avoid this bug.

This fixed in the current `master` branch.

LipuFei, 2019-09-03:
I think this is new in 4.0, for reference, the JIRA ticket is CURA-6011. It was added last December. https://github.com/Ultimaker/Uranium/pull/433

I will close this ticket now.

closed 2019-09-03 by LipuFei

Thisismydigitalself, 2019-09-04:
Thank you for your prompt help. Does this bug fix means next release will allow us users to use "[ ]" for profiles names?

LipuFei, 2019-09-04:
Yes, the fix will be in the next release. so the names you are using will be ok.