Ultimaker/Cura #6277: HORIZONTAL Expansion menu

ariefstenza, 2019-08-29:
**Application version**

Cura 4.2.1

**Platform**

Windows 7 Ultimate

**Printer**

Ender 3 Pro

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

The value of Horizontal Expansion cant show "-" Character, but the actual Print the setting could work properly

fieldOfView, 2019-08-29:
I can't reproduce this. The - can only be entered as the very first character in the field.
Oddly, in your screenshot, it looks like there is a space before the first number. I don't know how that could have gotten there; it should not be possible to enter a space in that field.

ariefstenza, 2019-08-29:
yes..i have try to enter - first but the character not show

i do that on Cura 4.0 the - could show

fieldOfView, 2019-08-29:
How about other fields that should be able to accept negative values? Eg Retraction Distance?

ariefstenza, 2019-08-29:
cant too

This is attach setting Using CURA 4.0.0

[image: image.png]

diegopradogesto, 2019-08-30:
I also can't reproduce the issue. The "-" sign appears correctly. Is it maybe a rendering thing? Do you have an external monitor?

Ghostkeeper, 2019-09-03:
I'm more interested in a project file for this "Ender 3 pro" thing.

Ghostkeeper, 2019-09-26:
We've tested this on several computers and couldn't reproduce this. Though on one computer the minus sign was slightly blurry because of weird Windows font rendering. Perhaps for some DPI screens it disappears entirely?

fieldOfView, 2019-09-26:
For completeness sake, here is an image that shows a font rendering difference between 4.0 and 4.2/newer (provided by @Stormwing01):
[image]

Stormwing01, 2019-09-26:
Coolness. Hopefully the issue gets resolved :) Wouldn't have thought it's a font issue... how odd.

RegRayRat, 2019-10-24:
Thanks for redirect ( I did do a quick search before post so sorry I missed this thread.)
Just for completeness; I do see a movement of the cursor but not the  - sign.
I changed the screen resolution but to no avail.
I have downloaded Cura 4.3 to a laptop running Win 7 but still cannot see the - (Negative) sign.
I will try measuring test prints to see if the -0.05% has been applied.
And, of course will keep an eye on this thread -  weird bug indeed.

fieldOfView, 2019-10-24:
If you change an infill setting (so that the label becomes cursive), do you also have an undercase "f" that extends below the font baseline like in the screenshot posted above? I am convinced this is a font rendering issues, but I don't know why some people seem to have a completely different font rendering in the italics than others.

Do you by chance have the Noto Sans font installed as a system font?

Ghostkeeper, 2019-10-25:
I've seen this as well on my Windows 8 laptop now.

I'm beginning to see a pattern. Seems like this is not an issue in Windows 10 but only in 7 and 8 so far. Maybe it's got something to do with the ClearType implementation in Windows, because that changed a lot going to Windows 10.

fieldOfView, 2019-10-25:
The difference is more than just Cleartype. @Ghostkeeper could you confirm that the italic lowercase "_f_" renders with the "leg" of the "_f_" extending below the baseline of the font and curling to the left a bit like in the screenshot above? That means that Windows renders an entirely different set of glyphs for italics. 

On my system (and apparently on older versions of Cura as well), italic text renders as just regular text that is slanted. This is called "faux italic". The rendering in Windows 7 and 8 is actually more correct to what the font designer intended (though it is missing the minus sign).

Ghostkeeper, 2019-10-26:
The regular "f" does not extend below the base line, but the italic "f" indeed does. That means that it's actually using italics, not just oblique or "faux italic". I don't have Noto Sans installed on my computer, so this font must be coming from Cura.

Note that the setting text boxes in question don't contain italics or obliques. It's just regular.

In older Cura versions it wasn't properly including the Noto Sans font in the Windows build, which is why it looks very different. It was just using a system font then for most people.

cmarciales, 2019-11-05:
So what is the result of @fieldOfView and @Ghostkeeper's discussion? I'm using Win7 and cannot see the negative signs on any input window, including the "location", which is troublesome. If this is a Font issue, is there a way to change the Font in Cura so that I can see the negative signs? It was working just fine several versions ago (like before 4.1), but now it's a chore when trying to line up objects on the bed using the "location" input. Any information will be helpful.

fieldOfView, 2019-11-05:
You could try this:
* go to Help -> Show configuration folder
* in the folder named `themes`, create a folder named `system_font`
* in that folder, place this file (named theme.json);
https://gist.github.com/fieldOfView/3e94440bcc66acfd014dd2f71f6a4885
* restart Cura
* in the preferences, select the theme named "Default font"
* restart Cura again

Stormwing01, 2019-11-05:
> You could try this:
>
>     * go to Help -> Show configuration folder
>
>     * in the folder named `themes`, create a folder named `system_font`
>
>     * in that folder, place this file (named [...]

it worked for me :) though this means I cannot use the dark theme anymore :( any tips on how I can manipulate it so that this fix works on the dark theme as well?

fieldOfView, 2019-11-06:
Where the json file says ‘cura-light’, change that to ‘cura-dark’ using a texteditor.

Stormwing01, 2019-11-06:
> 
> 
> Where the json file says `cura-light`, change that to `cura-dark` using a texteditor.

That worked, thank you :)

cmarciales, 2019-11-06:
Thank you @fieldOfView!!!!! That worked like a champ! Just in time as I upgrade (finally) to Win10Pro by getting a new computer. But this one will still be around so, it's good that I can now see the negative signs. Again, much thanks!

LipuFei, 2020-02-05:
@fieldOfView Thanks for the detailed info on this!

PR #7068 (2020-02-05): Update fonts using Google Noto Sans Fonts
CURA-6964

Fixes #6277

Fonts are downloaded from https://www.google.com/get/noto/#sans-lgc with the "Condensed" fonts removed.
closed 2020-02-06 by nallath