Ultimaker/Cura #6280: Right click --> Crash

Knusperhas, 2019-08-29:
**Application Version**
4.2.1

**Platform**
Windows 7 Home Premium 64 Bit

**Printer**
Custom Printer Standard Profile

**Steps to Reproduce**
1. I can start Cura, slice and everything.
2. I make a right click or try to open the menu on top (File, Edit, View,...)

**Actual Results**
Cura crashes and Windows is closing the program. 

**Expected results**
The Pop Up menu from the right click should open ...
 or the File menu should open

**Additional Information**
The Slicer settings on the right have no problem
It's a new installed windows on an old laptop. 
All Drivers are Updated by Dell Update Software.
All Windows Updates installed. 
Open GL until Version 2.1 is supported.

(I had Cura 3.5.1 running on this machine before new installation of Windows.)

I tried to remove Cura and install the 3.5.1 but it's the same behavior, so i upgraded back to 4.2.1...

Also tried all without antivirus and in administrator mode.

I installed Java and adobe flash Player, but nothing changes a thing.
I hope you understand what i mean and can help me out. 
If you need further information please ask for it, i would try my very best.

[cura.cfg.txt](https://github.com/Ultimaker/Cura/files/3556900/cura.cfg.txt)
[DxDiag.txt](https://github.com/Ultimaker/Cura/files/3556921/DxDiag.txt)
[curaRoaming.zip.txt](https://github.com/Ultimaker/Cura/files/3556908/curaRoaming.zip.txt)
[cache.zip.txt](https://github.com/Ultimaker/Cura/files/3556909/cache.zip.txt)

LipuFei, 2019-09-03:
@Knusperhas The logs files unfortunately don't contain very useful information. Is it possible for you to create a crash dump when that happens? Could you email it to me l.fei@ultimaker.com? The crash dump file should be a `.DMP` file. Thanks.

Knusperhas, 2019-09-03:
I sent you an private email with .dmp file.

LipuFei, 2019-09-03:
It seems to have something to do with OpenGL. From your `DxDiag.txt` file, looks like you have an Intel graphics card. Do you have the latest Intel graphics driver installed? Could you also try using the compatibility mode? You can enable it via the preferences dialog (see screenshot below). And you also need to restart Cura.

[image]

Knusperhas, 2019-09-04:
I am using the latest Driver. I checked it a couple of times. 
Sorry but i can Not Open the settings window because it is in these top Menü, where Cura also Crashes. 
Is it possible to Change it in some config File?
I tried an openGL Test program and Out of it i got the Info that OpenGL until 2.1 is supported

LipuFei, 2019-09-04:
Yes, you can change it in the configuration file. You can find it in `C:\Users\<your-usernamer>\AppData\Roaming\cura\4.2\cura.cfg`. You can open it with a text editor and add the line below in the `[view]` section:

```
force_layer_view_compatibility_mode = True
```

[image]

Knusperhas, 2020-04-25:
It's working correctly in new Cura Version

closed 2020-04-25 by Knusperhas