Ultimaker/Cura #6282: [4.2.1] Supports for empty space generated

bassovm, 2019-08-30:
**Application version**

4.2.1
**Platform**

Win 10
**Printer**

[D_spool.3mf.zip](https://github.com/Ultimaker/Cura/files/3559497/D_spool.3mf.zip)
**Reproduction steps**

Slice with supports enabled and support horizontal expansion set to 0
**Actual results**

Supports generated inside the model, along the vertical wall, that do not support anything
[image]
**Expected results**

Supports are only generated for the parts that have overhang and actually touch something
[image]
**Additional information**

For this model i need supports for inner groove overhang
[image]

StefanoV1989, 2019-09-02:
Yes, i have the same problem with supports in 4.2.1 version.
Yesterday it tried to put supports to a wall, without creating a base ... directly in the air linked only to the wall.

Ghostkeeper, 2019-09-03:
Duplicate of https://github.com/Ultimaker/Cura/issues/4219

closed 2019-09-03 by Ghostkeeper