Ultimaker/Cura #6283: Printing temperture not changing

rodribike, 2019-08-30:
Had ABS selected, then changed to PLA (both materials are my materials) but printing temperature didnt change.

in previous versions this was ok

**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

rburema, 2019-09-03:
This _might_ be an interface issue, if I remember correctly, we fixed an issue like that (and I can reproduce your issue in 4.2.1, but not in current master).

Can you check the produced gcode to check if the temperature _really_ doesn't change, or if it's 'just' and interface (update) issue?

Ghostkeeper, 2019-09-04:
The interface problem occurred only for settings that had a formula in the quality or variant profiles but NOT in the definition profile. It was occurring because Cura only makes settings depend on each other according to the formulas in the definition file, not in the InstanceContainers. It was not fixed but worked around by adding formulas to the Creality definition file.

I don't think that's the problem here, because the printing temperature is modified directly rather than via a formula.

Ghostkeeper, 2019-09-04:
It'd be useful to have a project file in this case, as it contains the profiles in question. Also it would be useful to know which printer you're using, so that we can see if there's anything awry with its profiles. Please fill in the template!

cicinovec, 2019-10-09:
Hi,
I have the same problem with latest Cura 4.3
The temperature change was working properly until I made small temperature adjustment directly in the quality profile (instead of material profile).
The only workaround is to make new profile by copying the default Cura profile and changing again all the parameters.

It would be helpful to have a toggle switch to change the value betwee profile settings or material settings (the small chain on the left side of the parameter) or to change the parameter again when material changes.

**Application version**
4.0 and 4.3

**Platform**
Windows 10 Home

**Printer**
Custom FDM printer

**Reproduction steps**

1. Create own profile copying of default Cura profile
2. Switch material. Default print temperature is changing with the material.
3. Make small change on default print temperature
4. Switch between materials. Default print temperature is not changing anymore
 
**Actual results**
No change of Default print temperature

**Expected results**
Default print temperature should change to value stored in material profile

**Additional information**
[example_project.zip](https://github.com/Ultimaker/Cura/files/3707330/example_project.zip)

nallath, 2019-10-10:
Well, you can change the default printing temperature like that already (eg; go to the material preferences page and change it there). That way it's directly changed in the material and it will have the behavior that you expect.

cicinovec, 2019-10-10:
Yes, that's correct,I know it now. But it would be great to have some possibility to reverse the value instead of creating a copy of profile again and again and again compare and change all already changed values

Ghostkeeper, 2019-10-10:
That issue is a duplicate of https://github.com/Ultimaker/Cura/issues/6082

Since the original issue has not had its template filled in, I'll assume that it was the same issue as what Cicinovec is having and I'll close the issue as a duplicate.

closed 2019-10-10 by Ghostkeeper