Ultimaker/Cura #6285: Outer wall feed rate ignores wall speed setting

tomoinn, 2019-08-31:
**Application version**
4.2.1

**Platform**

```
Linux Ogre 5.0.0-25-generic #26~18.04.1-Ubuntu SMP Thu Aug 1 13:51:02 UTC 2019 x86_64 x86_64 x86_64 GNU/Linux
```
**Printer**
Ender 3

**Reproduction steps**
Print the attached cube, using a single wall thickness to calibrate flow rate

[Flow calibration cube.zip](https://github.com/Ultimaker/Cura/files/3562316/Flow.calibration.cube.zip)

Zip file includes the GCode generated from that project

**Actual results**
Outer wall speed in single thickness layers is 10mm/s. GCode for each layer looks like this, so it's not a firmware or printer setting:

```
;LAYER:147
;MESH:Object 1
G0 F9000 X100.22 Y119.78 Z17.76
;TYPE:WALL-OUTER
G1 F600 X100.22 Y95.22 E537.26845
G1 X124.78 Y95.22 E537.80759
G1 X124.78 Y119.78 E538.34672
G1 X100.22 Y119.78 E538.88585
;TIME_ELAPSED:1903.444317
```

**Expected results**

Outer wall should be printing at 50mm/s as specified in the profile, so I'd expect the feed rate to be 3000 in the generated GCode

**Additional information**

smartavionics, 2019-08-31:
It's because you have a min layer time specified and so Cura has slowed the walls down to try and increase the layer time. Looking at your profile, the wall print speed is 25 not 50 so even if you reduce (or disable) the min layer time, it still won't use 50mm/s for the walls. Hope this helps.

[image]

tomoinn, 2019-08-31:
Ah, that makes perfect sense, thank you! I'll close this issue.

closed 2019-08-31 by tomoinn

thelastWallE, 2019-10-05:
Thx. Was thinking why it won‘t print faster, too.
My Ender 3 is on klipper so i wanted to print a calibration tower for pressure advance.