Ultimaker/Cura #6287: Slicer adds extra 'top/bottom layer' in middle of model

jcjveraa, 2019-09-01:
**Application version**
4.2.1

**Platform**
Windows 10 
**Printer**
Anycubic mega i3

**Reproduction steps**
Slicing my model with some 'recessed' letters at any of the 'built in' settings.

**Actual results**
From layer 134 Slicer adds 5 'top/bottom' layers (= 0.5 mm) underneath where some 'recessed' lettering will be printed, and then from layer 139 fills over that same area and the rest of the top with another 0,5mm completely filled top/bottom layers, making the earlier layers disappear and seemingly add no real value over just printing infill. If it is for strengthening, while not needed, it should at least then use a high printing rate. 

**Expected results**
No extra top/bottom layers in the middle of the model. 
[extra_top_bottom.zip](https://github.com/Ultimaker/Cura/files/3563277/extra_top_bottom.zip)

Images showing **feedrate** (blue 20mm, orange 50mm)
[image]
[image]
[image]
[image]

smartavionics, 2019-09-01:
Your profile specifies 12(!) top layers so it's doing exactly what you asked for. If you count them you will see that there are 12 layers under the letters.

jcjveraa, 2019-09-01:
Ah :)
Makes some sense I suppose, being equal to the 1.2 mm wall thickness?
But why does it then not start with complete layers but rather tracing the
outline of the letters?

Note, forgot to mention, this is a fairly hypothetical case - I can print
the object without the lettering to get rid of the problem.

smartavionics, 2019-09-01:
> But why does it then not start with complete layers but rather tracing the
outline of the letters?

Because then the areas around the letters would have more than 12 layers (17 in your example).

jcjveraa, 2019-09-01:
Ok, case solved, but still it feels slightly unintuitive.

closed 2019-09-01 by jcjveraa