Ultimaker/Cura #6290: Cura 4.2.1 Suggestion: Layer view

Thisismydigitalself, 2019-09-02:
**Application version**
4.2.1

**Platform**
Win10/64

**Printer**
CFFFP

**Reproduction steps**
Scrub to first layer, change object rotation

**Actual results**
once done adjusting object orientation, cura slices and then layer selector jumps back to last/top layer.
this annoying when trying to lay a model flat on bed. each time we try, slicer jumps to top layer.

**Expected results**
keep layer view at the selected one. maybe add a button to jump to top/all layers visible.

**Additional information**

rburema, 2019-09-03:
Sorry, we've decided to defer this for now.

diegopradogesto, 2019-09-10:
Using the "Lay Flat" option or the new in 4.3 "Select face to bottom" could help you with aligning your model to the buildplate.