Ultimaker/Cura #6293: [4.2.1] Support generation issue

Darko-Bdn, 2019-09-03:
ello all.
I have a question about Cura behavior with support mesh. I am trying to prevent excessive supporting structures on tall prints by using sacrificial "plate" that would act as a support base. It seems like Cura doesn't understand what's my goal 📷1f609.png😉
I tried to turn off Drop Down Support Mesh through Per Model Settings, but it doesn't make any difference. Support overhang angle is set to 50, while sacrificial plates are at 60°.
Any suggestions?
 
Pictures are showing a proof of concept model, not a real one. Cura makes proper support only on the first sample, the next one is already not functioning anymore.
 
One of the members in FB group discovered that if the sacrificial plate extends beyond the overhang, Cura understands it ok. But as far as the logic goes, proper behaviour would be as it is now on the first modeled "L " shape - but on all the L's....
 
Support blockers are not helping, as well as Conical supports.
 
If any more details are needed, please let me know.
 
Thank you.

**Application version**
4.2.1

**Platform**
Windows 8.1

**Printer**
Ender 5
[CE5_L support test model.zip](https://github.com/Ultimaker/Cura/files/3567675/CE5_L.support.test.model.zip)

**Reproduction steps**
Create supports with attached test model.

**Actual results**
Cura gives correct supports only on the first "sacrificial" plate. 

**Expected results**
Supports should be not generated below sacrificial plates, and dropped to the bed where the sacrificial plate is missing below the overhang.

**Additional information**
 If "sacrificial plates" are as long as the overhangs (or longer), Cura creates correct supports - not being extended to the floor. When "sacrificial plate" is covering the overhang only partially, results are not correct. Creation of correct supports only on one of the sacrificial plates is not correct.
[cura.log](https://github.com/Ultimaker/Cura/files/3567736/cura.log)

rburema, 2019-09-03:
As a workaround: Have you tried tree support? (Disable the 'normal' support, and enable 'Tree Support' in the experimental settings.)

Darko-Bdn, 2019-09-03:
Not at this model, but at another. Its not working well because Cura tries to support the overhang completely with the tree-like support from the ground level....

Ghostkeeper, 2019-09-05:
Looks like this issue again: https://github.com/Ultimaker/Cura/issues/4583

Darko-Bdn, 2019-09-05:
Oh yes it certainly does!

Ghostkeeper, 2019-09-06:
After some investigation, I discovered that this was not the same as #4583.

This bug is caused by the Support Join Distance setting. If that gets set to 0, the support generates as you'd expect.

What happens is that on both sides of the sacrificial plate there is support. Since the plate is so thin, the support on both sides is very close together. If it's closer than the Support Join Distance, the support gets merged together. This way the support can crawl underneath the sacrificial plate.

I'm not sure how to properly fix this at the moment though. Ideas are welcome.

nonloop, 2019-09-14:
[image]
[image]

Ghostkeeper, 2019-09-16:
@nonloop Those are nice pictures! I love dark mode. Did you have a bug to report? Please refer to the [bug reports](https://github.com/Ultimaker/Cura/issues/new/choose) button if you have a new bug.

Ghostkeeper, 2019-09-16:
We haven't found a solution yet to the issue of support passing through the model due to the join distance. We've considered doing an incremental closing operation for the join, but that makes support generation take like twice as much time (with like 4 increments). There is no easy solution for this.

Due to the ratio of how difficult it is vs. how often it occurs, and since it occurs only with an edge case, we're deferring this for later.

Darko-Bdn, 2019-09-16:
I can easily understand your decision. Still, like some other solution that you are puting under "Experimental" mode, this might also be left to the user's choice: pick "simple" supports, or turn detailed supports on (twice the time!). I am working on 10 years old PC that needs a couple of minutes to generate 52MB big g-code file. The time might not be such an critical issue here, but still could be let to the user's choice...

closed 2023-06-24 by github-actions[bot]