Ultimaker/Cura #6294: [4.2.1] Additional support bases - sacrificial plates added programatically

Darko-Bdn, 2019-09-03:
**Is your feature request related to a problem? Please describe.**
Creation of supports for overhangs on taller models always start from the floor level, except where Cura can find a piece of model beneath the overhang. On taller models this easily results in more material being spent on supports than on the model itself.

**Describe the solution you'd like**
A better way would be if Cura could find, define and place additional support bases (printed at angle with no need for supports) connected to the model to reduce support height (from below up). Connection of this additional support bases to the model itself should be designed to be easily broken off, like supports do already. That would make significat savings of support material on tall models, and consequently printing time.

**Describe alternatives you've considered**
Currently I am placing these "sacrificial plates" manually: just beneath the overhang, sloped down at 60°, extended to touch the model. On one specific model I've reduced overall material consumption and total printing time for about 30% (from ~350g to ~240g, for 110g netto weight of the model!). And I need to print 60 of them, that will be quite a saving. I believe Cura could place these plates quicker and in a more reliable way than manually.
 
**Affected users and/or printers**
This is not connected to the printer brand, more to the shape of the model: taller the model with overhangs, more of the savings achievable.

Ghostkeeper, 2019-09-05:
I was sure that we've been asked this before but I can't find it.

Algorithmically this would be pretty complex, though it can save a lot of printing time. We'll evaluate together with the team whether we see any time being put into researching this.

Darko-Bdn, 2019-09-05:
Or, I have possible another approach: when you turn off supports, but turn on "make supports printable", Cura tries to make additional geometry that (under "printable" angle) additionally "reinforces" geometry of overhangs in a way to enable them being printed. That functionality should be improved: see attached stl: overhanging horizontal bars get a triangular part beneath, but it should be sloped also towards the vertical part of the model. If this reinforcement would actually become a support which would not start from the ground level, that would be ideal solution!

Darko-Bdn, 2019-09-05:
Oh, the stl...
[L structure.zip](https://github.com/Ultimaker/Cura/files/3581111/L.structure.zip)

Ghostkeeper, 2019-09-06:
Yeah, the "Make Overhang Printable" setting is in experimental for that reason. It can make the object thinner in every direction but it is not able to do the path planning routine required to prevent getting overhanging points or lines. Currently it basically works the same as conical support but with the model instead of with support and without having a "minimum diameter" setting.

We've talked about the proposed changes here. One idea that seems a bit better is to have a way to make conical support grow towards the model in the same way as what tree support does when there is no available path to the build plate. However with tree support it grows toward the model by moving 2D points towards the nearest polygon, and then later drawing circles around those points. With area support, this is a lot more complex. Moving the vertices towards the nearest polygon wouldn't be desirable. It would need some sort of iterative approach with a bunch of opening and difference operations to maintain some sort of minimum polygon width while shrinking in every direction.

Additionally there would need to be some mechanism to temporarily ignore the X/Y distance in order to attach to the model, with a configurable setting to determine how long the X/Y distance needs to be ignored.

We've decided to defer this for now. It's likely that it won't make as much of a difference in material usage as tree support, and seeing the difficulty in implementation, it's risky to invest too much resources in it now. It may be a good research topic for later though.

closed 2023-07-13 by github-actions[bot]