Ultimaker/Cura #6299: [4.2.1] Invisible dragging bars in the machine settings

Liger0, 2019-09-03:
**Application version**
4.2.1

**Platform**
Win10 x64

**Printer**
Any

**Reproduction steps**
Try to move the bars on the gcode fields

**Actual results**
Bars are not visible before you drag them, and barely visible while selected.

**Expected results**
Bars visible

**Additional information**
https://youtu.be/MMqV_Un639I

Ghostkeeper, 2019-09-05:
This is the default Qt behaviour with Qt Controls 2. I'm not sure how modifiable this is with the v2 of Qt Controls.

The same happens with basically all of our new dialogues. For instance with the change log (Help -> What's New).

nallath, 2019-09-05:
It is modifiable. We can define our own scrollbars.

diegopradogesto, 2019-09-09:
I did some work a while ago in order to change the scrollbars to Qt Controls 2 and also align the styles in all the places where a scrollbar appears. I have rescued it and I think it could improve this request.

I'm going to link the PR to our internal ticket.

mahtDFR, 2020-04-28:
Internal ticket here: CURA-6763

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

closed 2023-06-24 by github-actions[bot]