Ultimaker/Cura #6303: Per model settings and cubic subdivision, possible bug CURA 4.2.1

Queneri, 2019-09-04:
**4.2.1**

**Windows 10 and MacOs Mojave**

**Ultimaker 3 extended**

**Opening several files. Selecting per model settings, changing the infill of one of those files applies the change to all **

**Apply the change to all models**

**Apply the change to the selected file**

** only happens when selecting cubic subdivision infill pettern. **

Queneri, 2019-09-04:
Per model settings produces an error when cubic subdivision is the general infill parameter. When attempting to change the infill parameter of a single model, it applies to all remaining models. 
another result that you can expect when processing the model is 0% infill
[image]
[image]

Queneri, 2019-09-04:
Playing a little more with the problem, I found another result. This time the infill is cut and leaves hollow parts inside the model
[image]
[image]
[image]
[image]
[image]

rburema, 2019-09-05:
I can reproduce this on an UM3, I'll discuss it with the others.

tecnocat, 2022-01-20:
Cura 4.7.1 and Windows 10 with same issues.

General infill set to 0% and Gyroid, and each box of model infill with all of them, every infill per model works perfectly except the Cubic Subdivision:

[image]
[image]

tecnocat, 2022-01-20:
I uploaded the project to test by yourself, GitHub block the file type so rename to .3mf first.

[cura-4.7.1-infill-patterns-3mf-file.txt](https://github.com/Ultimaker/Cura/files/7908262/cura-4.7.1-infill-patterns-3mf-file.txt)

Ghostkeeper, 2022-01-27:
Your issue is different though, @tecnocat. The original issue is that changing the infill of one object to Cubic Subdivision causes all models to use Cubic Subdivision, and that's clearly not the case for you.

closed 2024-12-07 by github-actions[bot]