Ultimaker/Cura #6304: Show start and end gcode movements and extrusion

Liger0, 2019-09-04:
**Is your feature request related to a problem? Please describe.**
Not much of an issue but as of now it's hard to tell if the modell will intersect purge lines or troubleshoots gcode to see if the code you wrote is effective or not.

**Describe the solution you'd like**
Let cura show the movement and extrusion actions in the preview that are defined as start and ending gcode. For example S3D does it.

**Describe alternatives you've considered**
None

**Affected users and/or printers**
Any

**Additional context**

rburema, 2019-09-05:
The preview right now is not really dependant on the gcode. In order to build this, we'd have to build the entire gcode analysis as well... So it might be a bit more effort than you'd think!

Ghostkeeper, 2019-09-06:
It would need to parse the g-code, which takes pretty long. You can see this when you load a g-code from file. It can take a minute.

We're at a disadvantage here because our front end is based on Python which doesn't perform well in these situations. If we were to implement this, the g-code parsing and producing Numpy data from it would need to be brought over to a SIP module.

It's impossible to parse just the start g-code, since the start g-code will influence the main g-code too (e.g. with `G92`). Or at least it will display inaccurate results, then.

fieldOfView, 2019-09-06:
> It's impossible to parse just the start g-code, since the start g-code will influence the main g-code too

I don't understand. Can you elaborate?

Ghostkeeper, 2019-09-06:
If the start g-code does this:

    G0 X10 Y0
    G92 X0

Then the entire model shifts 10mm to the right. More realistically:

    G92 E0
    G1 E-6.5

This would cause subsequent travel moves to be retracted. But Cura would not display it that way.

Of course that is currently also an issue, but if the start g-code is rendered we'd raise expectations that it actually works.

Thisismydigitalself, 2019-09-10:
https://github.com/Ultimaker/Cura/issues/3128

oo-dev17, 2023-04-04:
@Liger0 are you aware of the workaround to save the gcode to a file and load (open) it back to cura to see the start code movements?