Ultimaker/Cura #6306: [4.2.1] Jerk control not disabled after un-checking the checkbox in the interface.

daaz68, 2019-09-05:
**4.2.1**

**Windows 7 64 bits**

**Custom made delta printer**

**Reproduction steps**

I configured and played around with jerk control values and later on decided that I want to disable the feature.

**Actual results**

After un-checking the "Enable Jerk Control" checkbox, I noticed that the printer behavior did not change back, therefore I looked at the configuration file and noticed that the  "jerk_enabled = False" is missing and the last settings for the jerk control were still in there.

**Expected results**

Jerk control disabled in the behavior of the slicer, as per the GUI settings.

**Additional information**

After I manually edited the configuration file, the jerk control was finally disabled and the slicer went back to the jerkless behavior.

rburema, 2019-09-05:
Could you try the following?: If you save the project when this happens, close it, then open again, what is the value of the jerk-control in the GUI?

daaz68, 2019-09-05:
Tried as instructed. The printer configuration is updated correctly, with no jerk control information. Moreover, when updating the settings with the settings taken from the project file, all settings are recorded correctly, with and without jerk-control and values.

My scenario did not involve using a project, instead I was just saving and modifying the printer profile settings. My bad, I forgot to mention this.

Ghostkeeper, 2019-09-06:
I'm not able to reproduce this. Not entirely at least. Here are my reproduce steps.

1. Add DeltaBot printer.
2. Load a cube.
3. Enable Jerk control.
4. Change Infill Jerk to 42mm/s.
5. Disable Jerk control.
6. Save g-code and look for M205 commands.

I don't find any M205 commands, so it seems like Jerk is not being controlled by the g-code any more once the setting is disabled. At the end of the g-code file it also properly mentions that `jerk_enabled = False`.

I do notice that something weird is happening though: For jerk and acceleration, it doesn't remove the reset arrow (or the setting override) when it's back on the default. Maybe that has got something to do with those settings being global rather than per-extruder.

daaz68, 2019-09-09:
Got it; I will try and use your steps and check for the result.

Thank you, a lot!

Yours,
Daniel

closed 2020-06-19 by no-response[bot]