Ultimaker/Cura #6309: [4.2.1] After opening a Gcode and deleting, slicing will not work

anonimou0, 2019-09-05:
**Application Version**
4.2.1

**Platform**
Linux AppImage

**Steps to Reproduce**
1. Start Cura
2. Open a *.gcode file
3. Delete the *.gcode file (CTRL+D)
4. Open a *.stl file
5. Slicing is not available

**Actual Results**
Slicing is not available until starting a new project or restarting Cura. This is bad because the current settings will be lost.

**Expected results**
Slicing is available.

**Additional Information**
Note that starting a new project, without restarting Cura, will make slicing for *.stl available again. But then, the current settings will be lost.

smartavionics, 2019-09-05:
Yes, this bug has been around for some time. See https://github.com/Ultimaker/Cura/issues/5089.

anonimou0, 2019-09-06:
Yeah, this seems to be the same bug. That bug however is marked as Needs info. Anyway, fell free to close this one if you want to keep just that.

Ghostkeeper, 2019-09-06:
Yeah, closing as duplicate of #5089. We need info on that bug because none of us have been able to reproduce the bug. We don't know what's going wrong. But it's also hard to tell what info we'd need.

closed 2019-09-06 by Ghostkeeper