Ultimaker/Cura #6312: cura stops as soon as you click open file

casper76, 2019-09-05:
I just downloaded cura 4.2.1 and every time I click open file it say cura exe has stopped working 
Processing an issue will go much faster when this is filled out, and issues which do not use this template WILL BE REMOVED and no fix will be considered!

Before filing, PLEASE check if the issue already exists (either open or closed) by using the search bar on the issues page. If it does, comment there. Even if it's closed, we can reopen it based on your comment.

Also, please note the application version in the title of the issue. For example: "[3.2.1] Cannot connect to 3rd-party printer". Please do NOT write things like "Request:" or "[BUG]" in the title; this is what labels are for.

It is also helpful to attach a project (.3mf or .curaproject) file and Cura log file so we can debug issues quicker. Information about how to find the log file can be found at https://github.com/Ultimaker/Cura#logging-issues 

To upload a project, try changing the extension to e.g. .curaproject.3mf.zip so that GitHub accepts uploading the file. Otherwise, we recommend http://wetransfer.com, but other file hosts like Google Drive or Dropbox work well too.

Thank you for using Cura!
-->

**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

nallath, 2019-09-06:
Could you please fill in the entire template?

Ghostkeeper, 2019-09-06:
Duplicate of https://github.com/Ultimaker/Cura/issues/5937. Try without any Windows shell extensions. This is an open Qt bug.

closed 2019-09-06 by Ghostkeeper