Ultimaker/Cura #6314: cann't load gcode file to print

lyfbuaa, 2019-09-06:
now only view gcode file

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Affected users and/or printers**
 
**Additional context**

Ghostkeeper, 2019-09-06:
I've not been able to reproduce this. These were my reproduction steps:

1. Add Ultimaker 3 printer.
2. Load a model file (1cm cube in my case).
3. Save to g-code.
4. Restart Cura.
5. Connect printer via wifi to my physical printer.
6. Load previously saved g-code file.
7. Click on Print via Network.

The printer starts printing that cube then.

I have a couple of questions for you. What are your...

**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

lyfbuaa, 2019-09-07:
1.open cura 4.2.1 in pc
2.load .gcode file
3.no button to print

Ghostkeeper, 2019-09-09:
I'm still not able to reproduce that!

I opened Cura 4.2.1, added an Ultimaker S5 printer (because otherwise it won't let me open any .gcode file) and then dragged the .gcode file onto the build plate. I see a button to print:

[image]

lyfbuaa, 2019-09-10:
sorry,I misreport it.
just I try again,the print button is there.

Ghostkeeper, 2019-09-10:
All right, so if the bug can't be found again I'll close it.

closed 2019-09-10 by Ghostkeeper