Ultimaker/Cura #6316: want extruder control function in monitor

lyfbuaa, 2019-09-06:
not only the extruder temperature control

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Affected users and/or printers**
 
**Additional context**

fieldOfView, 2019-09-06:
want more words to describe what you mean

Ghostkeeper, 2019-09-06:
Via USB cable you mean?

You can control the extruder. There are buttons to move the print head or the extruder. You can even send arbitrary g-code over the USB cable. This is already implemented.

fieldOfView, 2019-09-06:
There are no buttons to extrude/retract material. I can only guess that that is what OP is referring to.

lyfbuaa, 2019-09-07:
yes

Ghostkeeper, 2019-09-09:
Ah, sorry I misremembered. Those buttons to the side were for the Z axis.

fieldOfView, 2019-09-09:
The reason I did not add the extrude/retract buttons when I added the nozzle preheat functionality is that there is no way to extrude/retract that does not change the state of the printer (ie: you have to switch the active extruder). There is no way to detect the currently active extruder.

lyfbuaa, 2019-09-09:
let user choose or only the extruder 0 ?

fieldOfView, 2019-09-09:
Solutions that only work for single extrusion will generally not be accepted in Cura. 

Letting the user choose; how would that look? Can you draw me a mockup picture?
When switching active extruders, should the extruder start gcode snippets be run? Cura does not know what extruder is the currently active extruder (because the printer does not say so). What happens if you switch to the extruder that is already the currently active extruder and the start gcode snippet is ran again. Will that screw things up?

Ghostkeeper, 2019-09-09:
I don't think executing the start g-code is desirable. That often has routines in it like priming or wiping the nozzle that you wouldn't want to happen in a manual control workflow like this.

The printer firmware of the Ultimaker 3 allows moving filament, but when you first start moving it'll automatically heat up the nozzle. Of course the printer knows then which material is loaded in each nozzle so it knows what temperature to heat up towards.

Some printers can only move the filament for the active extruder though. It becomes a lot more complex then. Like fieldOfView said, Cura cannot know via USB cable which of the extruders is active. This is necessary for printers with mixing nozzles and direct drive feeders.

lyfbuaa, 2019-09-09:
I use the diy printer with marlin.when i change the material,i often use the funtion.
when I use the funtion in the repetier ,it require the preheat .   so I think if the preheated extruder side can be the funtion button?

GregValiant, 2024-11-24:
I'm going through and cleaning house.
Is this feature request still valid? or can it be closed?

closed 2024-12-09 by github-actions[bot]