Ultimaker/Cura #6318: Measure distance tool

classicstyleatx, 2019-09-06:
**What's the problem?**
measuring sizes of sub-components within an object is difficult to do solely by the X/Y grid on the print bed. Sometimes files are mis-labeled with incorrect sizing information, so the ability to measure within the application would be very helpful.

**Describe the solution you'd like**
A tool that allows you to draw a line between two points in space, _only in orthographic views_. This allows you to measure the size of a sub-component quickly to verify file data is accurate. Accuracy limited to hundredths of a mm or less.

**Describe alternatives you've considered**
Current alternatives are to load the part into a different piece of software, which slows down workflows and is prone to error due t loading the wrong file.

**Affected users and/or printers**
Anyone who designs and prints 3D printer parts or other mechanical parts is likely to benefit from this.

Thisismydigitalself, 2019-09-06:
You might want to read my feature request from two years ago.
https://github.com/Ultimaker/Cura/issues/2545

Ghostkeeper, 2019-09-09:
Yup, duplicate of https://github.com/Ultimaker/Cura/issues/2545.

closed 2019-09-09 by Ghostkeeper