Ultimaker/Cura #6319: Pop-up when hovering over info is not displaying correctly

mboratko, 2019-09-07:
**Application version**
4.2.1

**Platform**
Arch Linux

**Printer**
Creality Ender-3

**Reproduction steps**
After slicing, I place the cursor over the "info" icon on the bottom right.

**Actual results**
The text is displayed but it is all jammed together and runs off the screen. It looks like a pop-up window is trying to display but it is not working.

**Expected results**
Information displayed in a pop-up.

**Additional information**
[image]

smartavionics, 2019-09-07:
Hello @mboratko , are you using an AppImage or some other release? I ask because I know that the Arch people make their own releases which use later versions of Qt that cause some little glitches like you show above. The next release of Cura should have better compatibility with recent Qt (5.13).

mboratko, 2019-09-08:
@smartavionics you are correct, I was using [the version from the AUR](https://www.archlinux.org/packages/community/any/cura/), when I installed the AppImage from Cura's website the popup displayed correctly. Thanks for the quick diagnosis!

closed 2019-09-08 by mboratko