Ultimaker/Cura #6322: Variable layer height printing

BrianOz500, 2019-09-08:
Cura now allows for the layer height of the infill to be different to the wall layer height. This appears to work well.

Would it be possible for there to be a Cura setting that allowed for the surface print line of the wall to be a different layer height compared with the rest of the wall?

As an example the surface layer height could be 0.12 mm while the rest of the wall layer height and the infill layer height could be 0.24 mm.

This would give the benefit of a better surface finish without significantly increasing print time.

Ghostkeeper, 2019-09-09:
We've considered this before. It wouldn't work as well for walls.

Take for instance a pyramid with a slope of 45 degrees. Let's say we slice at 0.3mm layer height (so Z=0.15, 0.45, 0.75, etc) for the inner walls, but for the outer wall it slices at 0.1mm layer height, giving us 2 extra cross sections, one slightly above and the other slightly below the original slice (so Z=0.05, 0.15, 0.25, 0.35, 0.45, 0.55, etc). Let's make all line widths the default 0.4mm.

In our sloped wall, the inner walls slice at Z=0.15mm. It leaves 0.4mm spacing for the outer wall at that height. The outer wall needs to print within that spacing for all 3 (outer wall-) layers, at Z=0.05, Z=0.15 and Z=0.25. For Z=0.15 there is indeed 0.4mm spacing for the outer wall, so that one works fine. But at Z=0.05 there will be more space due to the slope of the pyramid: 0.67mm. At Z=0.25 there will be less space: only 0.13mm. So now the outer wall has to print at a line width of 0.13mm, which is going to fail pretty badly.

For infill this works fine, since the infill area is usually big enough that it won't fail. For walls it will almost always fail if the surface is sloped, which happens to be precisely the use case that you intended this feature for.

mimmori, 2019-09-25:
I believe that this function can be of great help in cases where the object does not present large slopes in the walls. It would make it possible to shorten printing times considerably.

Obviously where the wall has a slope it can be accepted to have a step imposed by the thickness of the inner layer even if the external layer could potentially be more precise.

Obviously the user will decide when it is appropriate to use it or not

BrianOz500, 2019-09-26:
I have carried out some research on the issue related to using variable wall heights.
The technique would require some rules for it to be effective. An example of a rule would be that the each successive layer height for each printed wall line from the surface would be twice the previous layer height. Also the wall would be printed from the outside first. The use of over-extrusion and small print displacements towards the previous printed wall line would eliminate the "holes" in the wall upto wall angles of about 45 degrees. I can post details of how this would work if anybody is interested.
One of the issue with 3d FDM printing is the layer lines that are left on objects. These layer lines can be covered with layers of paint, however, there are some instances where it would be useful if an acceptable surface could be produced without painting the object. I used a rather basic 3d printer is test whether or not it was possible to produce a skin layer of 0.04 mm on a 3d printed object. I chose this value since the diffraction limit of the human eye is about this value and even a basic 3d printer should be able to print at this layer height. The result of this experiment was that a matte finish was produced without obvious observable layer lines. Printing at a layer height of 0.04 mm for an entire object would typically be considered unacceptable in terms of the time taken, however, if only the outer surface is printed at 0.04 mm with each successive printed wall line from the surface printing at twice the previous height, then the time to print an object is not much greater than printing the entire object at a layer height of say 0.16 mm. Some people consider the matte surface finish produced with this technique to be quite attractive. I can attach some photos of some shapes that I printed with with a layer height of 0.04 mm if anybody is interested.

mimmori, 2019-09-26:
I can imagine the visual cleanliness of the object, but to see a picture would make everything closer to reality!
We hope to be able to afford this quality quickly without having a high cost in terms of time.

BrianOz500, 2019-09-29:
The attached image shows "1/2 height cube" blocks that I printed out on a four year old generic printer.
The top of the blocks have holes due to not having adequate thickness. This does not affect the quality of the printed walls.
The layer height for the walls was 0.04mm, the nozzle was 0.4mm diameter and the wall thickness was 0.8mm whale the infill layer height was 0.32mm without an infill wall.
The photo was taken from a distance of about 20 cm from the blocks.
The layer lines are not apparent at a distance of 20 cm and are not obvious or even observable at closer distances. The slight vertical lines are probably due to the quality of the printer.
The surface is not shiny like a standard 3d printed surface with a matte appearance which can be quite attractive, like a matte finish paint.
The ability to be able to produce a 3d printed part that has an aesthetically attractive surface without any post treatment such as painting would be useful in a number of applications.
This effect is similar to the retina display effect that is used in smartphone displays.

[image]

Elonsan, 2019-12-12:
> In our sloped wall, the inner walls slice at Z=0.15mm. It leaves 0.4mm spacing for the outer wall at that height. The outer wall needs to print within that spacing for all 3 (outer wall-) layers, at [...]

This is easily solved by just moving the inner walls in a bit. Sure, it might leave a small air pocket (which can possibly be solved by either over extruding a little bit). Point is, this is still doable. Have you tried it and the results looked poorly or is this all just theory? It's definitely worth trying since it can improve print quality a lot while keeping the time down :)

Ghostkeeper, 2019-12-12:
You can't move the inner wall sideways because it shares its extrusion path with a layer below it, where the lines can be shifted. So you'd have the walls being disjointed from each other, which makes them much weaker. You can make the adjacent lines wider but there is a limit to how much a line width can vary from the nozzle size. And there is the problem that real models don't have the same slope everywhere, so you'd have line widths that vary a lot around the contour, so the extrusion rate would have to adjust all the time and there's a delay there, so the print quality will suffer from that.

This is all theory, so far. I haven't implemented this. The aforementioned reasons also make it hard to implement. If the walls are sloped, how do you combine multiple layers of walls? There are no two contours that are the same on both layers. Multiple contours come together and split away across the layers. What if part of the wall suddenly jumps away because of a protrusion in the 3D model in certain layers? Do you interrupt the flow halfway a wall, or reduce the layer height? I'd rather not spend the time on implementing this until there is a decent plan of action.

GregValiant, 2026-09-10:
There is a plugin in the MarketPlace called "HellaFusion" that can change profiles at different layer heights.  You can change layer heights or other settings so long as they are in the particular saved profile.  There is a save option and you could even change nozzles.  Make no mistake - it is an advanced script and requires some practice.
I'll go ahead and close this.

closed 2026-09-10 by GregValiant