Ultimaker/Cura #6324: Question - Issue Skirt/brim arc radius, curve angle?

tablatronix, 2019-09-08:
4.2.1 OSX
Ender 3 pro

I am having issues onmy ender 3 pro printing skirts..

Is it possible to change the arc radius on skirts ?

For some reason my ender gets all jerky and cannot seem to print these well.. NO idea why, I have not checked for changes or reverted to an older version to see if this is new or not, never had this issue on my mpsmv2

[image]

I have video

Ghostkeeper, 2019-09-09:
With the Skirt Distance setting?

tablatronix, 2019-09-09:
tried 10 and 5

tablatronix, 2019-09-09:
vid https://youtu.be/P6iAfghjX7A

Ghostkeeper, 2019-09-09:
I suggest increasing the Jerk for skirt and brim.

The skirt's arc radius is exactly the distance between the skirt and the model, so that's worth a try as well.

Liger0, 2019-09-10:
Same on my cr10s. I don't have this on s3d and I don't have jerk/ascceleration control in cura.

tablatronix, 2019-09-10:
ok so I tested again with jerk and acceleration control and it is much better.
I am going to try to drop the skirt acceleration and see what that does, maybe its too high.

Ghostkeeper, 2019-09-11:
I think the default jerk of your printer is set very low. This means that it's not allowed to make a corner without slowing down significantly.

I've done a write-up of what jerk is exactly here in the Settings Guide plug-in: https://github.com/Ghostkeeper/SettingsGuide/blob/master/resources/articles/jerk_enabled.md

closed 2019-09-11 by Ghostkeeper

tablatronix, 2019-09-11:
Thanks, I thought I understood it apparently not, going to read some more, great write up.

Liger0, 2019-09-11:
I have a jerk of 8 in my firmware and in S3d have no issue.

Ghostkeeper, 2019-09-12:
Does Simplify3D put any M205 g-codes in its output?