Ultimaker/Cura #6331: Post Processing Cura 4.2.1

Firemann59, 2019-09-10:
**Application version**
4.2.1

**Platform**
Windows 10

**Printer**
Creality CR10s

**Reproduction steps**
Tried to use post processing: pause at height

**Actual results**
A comment was inserted into the gcode stating post processed but there was no other gcode modification.

**Expected results**
Gcode should have been modified

**Additional information**
I loaded a stl file, selected the profile I intended to use, selected extensions, post processing. I selected pause at height and filled in the parameters. I sliced the object and saved the gcode file to my computer. I then opened the gcode file with notepad and searched to find the gcode modification. It was not there. I then tried the same procedure with version 4.0 and it worked correctly.

rburema, 2019-09-11:
I've traced the reproduce steps in 4.2.1 and I _did_ get the `;POSTPROCESSED` (in the header) and the `;TYPE:CUSTOM` (&c) (at the site). (i.e. I couldn't reproduce.)

Could you attach the project file? (Zip it to get github to accept.)

P.S. I did observe that we've got 2 'pause at height' scripts now, one at the top of the list marked as 'for BQ printers only' and a more generic one in the middle of the list. It can't hurt to ask: Did you select the right script?

cobaltjeh, 2019-09-18:
I am having the same problem, cura 4.2.1 is not applying any post processing when saving a file.

Had to revert back to version 4.0.0 to get it working again.

Using windows 10 + Creality Ender 3 printer.

rburema, 2019-09-18:
@cobaltjeh Could you attach the project file? (Zip it to get github to accept.)

cobaltjeh, 2019-09-18:
I re-installed version 4.2.1 just to be sure, in the zip the project file is included, and the stl +generated gcode.

Layer pause should be at 0.35, 0.20 initial layer and 0.15 for the rest of the layers. Gcode doesn't show anything. While this was working in version 4.0.0.
[CE3_PostProcessing_Test.zip](https://github.com/Ultimaker/Cura/files/3625371/CE3_PostProcessing_Test.zip)

rburema, 2019-09-18:
Thanks!

It seems there's something going on with loading the project: The specified script is present in the bowels of the .3mf file, but doesn't show up in the interface. If you then select another script, it saves over it.

cobaltjeh, 2019-09-18:
The project is loading fine here, the interface is also showing the post processing icon next to the slice button.

And the script is present when opening via the menu.

Firemann59, 2019-09-18:
I did not save the file and when I tried to re-create it it is now working. I have attached the gcode file that I did save and you can see it shows postprocessed but I can see no post processing code in the file.
[CCR10S_Mexican Train Hub-9-Pause.zip](https://github.com/Ultimaker/Cura/files/3627357/CCR10S_Mexican.Train.Hub-9-Pause.zip)

Ghostkeeper, 2019-09-19:
The project file does properly load the post-processing script for me.

I'm seeing the bug also. It's happening because it's the last layer. We moved the Z travel move to just before the layer switch and then it's skipping it somehow.

Ghostkeeper, 2019-10-22:
We've decided to defer this issue for now. We're not giving it any priority over our current plans, but we'll keep it open to track it and see if we can make time for it later if it proves to be a bigger problem.

cobaltjeh, 2019-10-22:
If it helps, the problem seems related to the z-hop setting. If I disable z-hop the pause on layer works as expected. With z-hop enabled the pause on layer doesn't work, or pauses at wrong layers.

Z-hop was enabled by default in Cura 4.2.1 for the ender 3 profile. And again disabled for the ender 3 profile in Cura 4.3.0. So that is why I was having problems with 4.2.1. I have replicated the problem in version 4.0.0 and 4.2.1. So I don't think it is version related.

closed 2023-06-24 by github-actions[bot]