Ultimaker/Cura #6333: expect more intuitive custom supports

lyfbuaa, 2019-09-11:
cura by custom supports plugin:
[image]

simplify3d:
[image]

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Describe alternatives you've considered**

**Affected users and/or printers**
 
**Additional context**

rburema, 2019-09-11:
Hi! (A lot of) Plugins aren't the responsibility of the people developing Cura (though there is significant overlap), you'll want to repost your request [at the issue tracker of the plugin](https://github.com/lokster/cura-custom-supports/issues) instead.

I'm sure though that when you do, they'd appreciate some more information: e.g. what do you think is unintuitive about their current approach?

Ghostkeeper, 2019-09-11:
Discussion is continued here: https://github.com/lokster/cura-custom-supports/issues/5

closed 2019-09-11 by Ghostkeeper