Ultimaker/Cura #6336: panning, zooming and rotating

pineirod, 2019-09-11:
To have a panning, zooming and rotating system like the one of Preform. When selecting a part it would be great to have a sphere or the rotating rings, as well as the bars for panning and zooming. Here a picture from Preform.

Also, click and drag to create a selecting rectangle in order to select different parts at the same time
[image]

rburema, 2019-09-12:
You can already select multiple parts with shift, panning with the right mouse button, and zoom with the middle-mouse scroll-wheel. Also, in preferences, you can have the current part be always centred for easier zoom.

Also, once a part (or parts) are selected, you can push the rotate-tool button (on the left side of the screen). Then, tool-handles similar to what is shown in the picture will appear. (Similar for the translate-tool, for dragging the object(s) everywhere.)

Does this help with your needs or is this more of an 'enhancement' request where you're dissatisfied with the current way those tools operate?

Block137, 2019-09-14:
I'd like something similar to Blender, press R to bring up rotation gimbal, press X or Y or Z to start rotating along corresponding axis, drag mouse around object center to rotate, left click to stop, esc to cancel rotation. S for scale and G (grab) for position.

BTW
Ctrl + A to select everything works

elmo40, 2019-09-21:
"... you're dissatisfied with the current way those tools operate?'
YES!
There is an input box for placement and scale yet nothing for rotation? How has this been overlooked? Repetier has it.
I do notice the 4.3 Beta feature with selecting a face and placing it parallel to the bed but that is not what I can use. Many parts need a certain area to not have a jagged surface and I need to set the angle for a flat top, tediously, with the mouse. Please add an input box for angle selection.

Ghostkeeper, 2019-09-24:
@elmo40 that is being discussed here: https://github.com/Ultimaker/Cura/issues/2779