Ultimaker/Cura #6337: translation to Czech

adekarem, 2019-09-11:
good evening,

I would like to ask you how to proceed if I would like to translate Ultimaker Cura into Czech.

I never translated programs, translated movies and dramas, documentaries, I would like to try. I have already found some information on the Internet, and I would therefore like to make sure that this is the right procedure. I found in the c: \ Program Files \ Ultimaker Cura \ resources \ i18n template folder to accurately translate 4 templates that can be opened and translated in poedit. I don't know how to proceed please

S474N, 2019-12-23:
> good evening,
>
> I would like to ask you how to proceed if I would like to translate Ultimaker Cura into Czech.
>
> I never translated programs, translated movies and dramas, documentaries, I would like [...]

Postoupila si nejak? Pripadne nabizim pomocnou ruku s prekladem!

adekarem, 2019-12-28:
Omlouvám se za pozdní odpověď něco přeloženého mám, ale momentálně z důvodu
nemocí jsem překlad pozastavila, mohu se zeptat zda víte jak to dát do kupy
poslala bych vám co mám a popřípadě bych něco i přeložila navíc děkuji za
odpověď

Ghostkeeper, 2019-12-31:
The most common way to share translation files is via .po files, if that's what you're asking. Google Translate is borking on your sentence a bit.

Ghostkeeper, 2020-02-28:
Someone else has recently translated the Cura repository to Czech. It just needs to also translate Uranium now before we can add it.

See https://github.com/Ultimaker/Cura/pull/7141

Ghostkeeper, 2020-05-07:
This is merged for Cura 4.6.

closed 2020-05-07 by Ghostkeeper