Ultimaker/Cura #6350: Speed visible by pointing in the legend bar

Liger0, 2019-09-13:
**Is your feature request related to a problem? Please describe.**
Understand what speed a color refers to is a real challenge, as there is only a min and a max value on the respective colour

**Describe the solution you'd like**
When pointing on the colour bar,  the horizontal coordinate of the cursor should be read so that it creates a popup with inside a speed value that tells what that color refers to.

**Describe alternatives you've considered**
Put a fixed little rectangle (like a manga balloon) that updates from a 0 value when hanging over the bar. It would look even cleaner.

**Affected users and/or printers**
Any

**Additional context**

smartavionics, 2019-09-14:
What would be far better would be to ditch the stupid nozzle graphic and show a balloon containing the current coordinates, speed, feedrate, etc. Just like a decent gcode viewer would. Take a look at craftware and you will see a good example of this. Maybe add a checkbox to enable this display.

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

fieldOfView, 2021-06-04:
> What would be far better would be to [...] show a balloon containing the current coordinates, speed, feedrate, etc. 

I started doing that here:
https://github.com/Ultimaker/Cura/pull/9036
I'll see if I can make a plugin out of that, once I have time for plugins again.

closed 2023-07-13 by github-actions[bot]