Ultimaker/Cura #6351: Time Estimation

Slider99, 2019-09-14:
**Is your feature request related to a problem? Please describe.**
 Very New to 3D printing, and I would like to see when you load a .gcode that it gives you the time estimation, If I have sliced it with Cura and it shows when slicing, why is that information not saved and the possibility to show as well?

**Describe the solution you'd like**
 When Saving please save all relevant information, as per when its first sliced.

**Describe alternatives you've considered**
 Only way around it at the moment is to save the gcode into a directory with file name and hours to complete, or save a .txt file with the information.

**Affected users and/or printers**
 I would expect everyone would benefit as this function doesn't seem to be there

rburema, 2019-09-17:
As a workaround, you can save (not export) to a project file. The downside is that you'd need to reslice every time, but all  the settings, object orientations, etc. etc. are saved.

Ghostkeeper, 2019-09-17:
As an aside, I've ported CuraEngine's time estimator to Python a while ago: https://github.com/Ultimaker/Cura/blob/master/scripts/check_gcode_buffer.py

Even if the g-code doesn't contain a time estimation comment, we could still estimate it (but it'd take a few seconds extra).

fvrmr, 2021-06-03:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.