Ultimaker/Cura #6356: [4.3 beta] Main Wipe retraction speed not inherit

Liger0, 2019-09-14:
**Application version**
4.3 beta

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
Enable wipe

**Actual results**
The main wipe retraction speed is unused by the profile, because both its submenus inherit the speed directly from the retraction speed instead of the upper field.

**Expected results**
The subvalues should inherit values from the upper value, which then inherits it by the material retraction speed

**Additional information**
I think it's been forgotten when implemented because even the description doesn't report it being an inherit value, but then it doesn't make sense to exist (and have a 5 value by default) if its value is never used.
[image]

rburema, 2019-09-17:
Seems like those inherit the value from 'retraction_speed' instead of 'wipe_retraction_speed'.

Ghostkeeper, 2019-09-17:
I think we should remove the parent setting and let it inherit from the retraction settings then. Maybe we could even put them as child settings of the retraction speed setting.

rburema, 2019-09-17:
See new Jira ticket: CURA-6789.

rburema, 2020-12-11:
Seems this was fixed in commit https://github.com/Ultimaker/Cura/commit/e2b8a44 by making the parent setting relevant again.

closed 2020-12-11 by rburema