Ultimaker/Cura #6357: [4.3 beta] Nozzle offset has no effect

Liger0, 2019-09-14:
**Application version**
4.3 beta

**Platform**
Win10 x64

**Printer**
Cr10S

**Reproduction steps**
I have my nozzle, when homing the Y, 20mm out of the bed.
I tried to compensate for that using a -20mm on the nozzle offset in Cura

**Actual results**
But it didn't make any difference in the coordinate.

**Expected results**
If I understand the offset function correctly, it should read the Y coordinates and sum 20 to them so making my nozzle 20mm ahead.

**Additional information**

rburema, 2019-09-17:
Sorry, I can't seem to reproduce this: my .gcode is different for 0mm and -20mm offsets when I try slicing with both.

Ghostkeeper, 2019-09-17:
Note that Cura's layer view compensates for the offset such that it'll display as if there is no offset (towards how the print is actually produced). To see the offset in a layer view, you'd need to load the g-code while a printer is active that has no nozzle offset, or otherwise load it in a different g-code viewer.

Liger0, 2019-09-17:
I tried to print the gcode to see that it makes no difference.

Ghostkeeper, 2019-09-18:
We haven't been able to reproduce this yet. Do you have a project file that reproduces the issue?

Ghostkeeper, 2019-09-19:
Yeah, the extruder start and end g-code are not modified except to fill in the setting variables. I think that's also important because it reduces the magic going on, which is usually desirable for a tinkerer's feature like the extruder start and end g-code.

It would also break existing printers that use the start and end g-code. And of course the main start and end g-code doesn't have any offset since it's not defined for any particular extruder but just global.

closed 2020-06-19 by no-response[bot]