Ultimaker/Cura #6358: Printer settings text is not visible.

philwrenn, 2019-09-15:
**Application version**
4.2.1 - Linux 64 AppImage

**Platform**
Linux Mint - nvidia-driver-390

**Printer**
Creality CR-10S

**Reproduction steps**
Open Cura, then open Machine Settings dialog.

**Actual results**
None of the text in the settings fields is visible.

**Expected results**
Settings values should be visible.

**Additional information**
[image]

philwrenn, 2019-09-15:
I discovered loading a custom theme with setting_control size set to `[11.0, 3.0]` makes the values visible and the controls usable.

```json
{
    "metadata": {
        "name": "My Theme",
        "inherits": "cura-light"
    },
    "sizes": {
        "setting_control": [11.0, 3.0]
    },
}
```
 Is there a way to set the default initial size of the Machine Settings dialog so the G-code text areas are visible without a window resize?

nallath, 2019-09-18:
Hmm, this is really weird. I've looked into this bit of code, but it seems we're just letting qtQuick take the wheel here. 
There were some warnings that i've fixed, but since I don't actually have the issue myself, it's impossible to say if this fixed it.
Would you be willing to test a nightly build to see if my changes fixed it?

Ghostkeeper, 2019-09-23:
We don't have our build system set-up to publish nightly releases on a public place yet. It's only on a server in Ultimaker's network right now. That's a TODO on us. However we are able to share the nightly releases that it builds (at least the ones that don't contain support for upcoming Ultimaker products yet). So here you go, the build of this morning:

http://dulek.net/work/Cura-4.4.99-master.20190923014613.AppImage

(Uploading right now; should take a few minutes.)

This is an unstable build of master though, evolved beyond the 4.3 build. But you should be able to check the Printer Settings dialogue.

fieldOfView, 2019-09-23:
> (at least the ones that don't contain support for upcoming Ultimaker products yet)

Are you sure that there is no information about upcoming Ultimaker products in that master build?

Ghostkeeper, 2019-09-23:
No there is. We already published some files with the 4.3 release. They are public here on Github and will be released with the stable build tomorrow to a wider audience.

There are more things that we're keeping unpublished but those are not in this build (or in 4.3) :wink:

philwrenn, 2019-09-23:
Printer settings dialog looks good in that build when using the default theme. That build also resolves quite a few other minor visual issues that I suspect were related to my display scaling.