Ultimaker/Cura #6361: [4.0.0] Infill Density Doesn´t Match (URGENT!)

Regiux, 2019-09-15:
**Application version**
4.0.0

**Platform**
Windows 10 Home - Nvidia GeForce GTX1050Ti

**Printer**
Creality Ender 3

**Reproduction steps**
Open Cura > Load Model > Slice and Preview

**Actual results**
Infill Density doesn´t match the resulted infill desired in the sliced model

**Expected results**
Infill Density designated for slicing should be the same as in the sliced model

**Additional information**
This issue only happens with some models
[image]

Liger0, 2019-09-15:
Without a project file I don't think anyone could help.

Regiux, 2019-09-15:
Added file download for project reference

smartavionics, 2019-09-15:
Sorry, that's not a project file it only contains the model but we need to see the settings. Please do File -> Save to produce a .3mf file, either rename or zip and attach to this issue. Thanks.

Regiux, 2019-09-16:
> Sorry, that's not a project file it only contains the model but we need to see the settings. Please do File -> Save to produce a .3mf file, either rename or zip and attach to this issue. Thanks.

Uploaded project file for possible study

smartavionics, 2019-09-16:
Thanks for the file. The reason that you don't get infill is because top layers is currently set to 999999! If you explicitly set top layers to something less, say 10, you get this...

[image]

Regiux, 2019-09-16:
Great Thanks !!!! This solved a lot for me!

closed 2019-09-16 by nallath