Ultimaker/Cura #6363: Printing nonexistent object

tphan26363, 2019-09-15:
Recently. the Cura 4.2.1 is having a problem that I can't figure out and don't know how to fix it.

It started printing outside the object at a a position 100 mm far way like a ghost image or some thing.  I have a video and a G-code file attached.  This is a huge problem for me since it create a lot of artifact grown out from the printing object.  A profile also included.  I look at the slicer but I don't see any problem from the preview
[image]

[Slicer.zip](https://github.com/Ultimaker/Cura/files/3614120/Slicer.zip)

[IMG_0638.zip](https://github.com/Ultimaker/Cura/files/3614118/IMG_0638.zip)

fieldOfView, 2019-09-16:
Check the "Wipe Nozzle Between Layers" setting.

Ghostkeeper, 2019-09-17:
Or rather, uncheck it!

closed 2019-09-17 by Ghostkeeper