Ultimaker/Cura #6364: [4.3.0 beta] Supports...unsupported

GMagician, 2019-09-15:
**Application version**
4.3.0 beta

**Platform**
Win10

**Printer**
Custom

**Reproduction steps**
Slice project and see preview

**Actual results**
Some supports (the right on image) has unsupported areas. The one on left is different (maybe not perfect) but piece is pice left and right are identical

**Expected results**
Identical support on identical overhanging parts and not missing parts under support

[image]
[PSU power connectors (New).zip](https://github.com/Ultimaker/Cura/files/3614238/PSU.power.connectors.New.zip)

Ghostkeeper, 2019-09-16:
I'm not able to reconstruct your exact case since you've modded Cura with a custom printer definition ("Prusa i3 Mk3s Clone"?) that I don't have. But when slicing in Custom FFF Printer with support enabled I see a similar thing.

This is just the result of the zigzag pattern being inconsistent in which side gets the connecting piece between two lines. Cura will try to put the connecting piece on the same side in every layer, but this isn't always possible because the area to fill with zigzags can grow and shrink in all directions, so sometimes lines appear and sometimes they'll disappear or disappear partially.

To prevent this there's a few options available:
* Tune the infill density so that this phenomenon happens to not occur any more.
* Tune conical support so that this phenomenon happens to not occur any more.
* Change the support pattern to something else, such as Lines.
* Add an extra wall around the support (Support Wall Line Count setting), possibly combined with the previous workaround.

Or you could just accept this. Chances are it'll just print fine.

We're closing this issue since it's accepted that the connecting pieces hang in mid-air. It's simply a geometric constraint.

closed 2019-09-16 by Ghostkeeper

GMagician, 2019-09-16:
ok, thanks for hints to solve it, I'll try them