Ultimaker/Cura #6365: Cura adds double *.extension when saving files.

enVide, 2019-09-16:
**Application version**
4.2.1

**Platform**
Windows 10 Pro English 64
Intel i-8700 CPU 3.2GHz HEXACORE
RAM 32 Gb
NVIDIA GeForce GTX 1060 3Gb

**Printer**
Creality Ender-3

**Reproduction steps**
Saving files as *.gcode or even the native *.3mf.

**Actual results**
The filename suggestion appears with a doubled extension (per example *.gcode.gcode)

**Expected results**
Single extension.

**Additional information**
When choosing to "Save as type:" the dropdown menu has doubled entries for each file format.

rburema, 2019-09-17:
Sorry, I can't seem to reproduce this. Does this still happen with 4.3-BETA?

horga83, 2019-11-25:
Cura 4.3.0 on Arch Linux still adds a double .gcode to the gcode file so one ends up with  blah.gcode.gcode as an example.

ve7it, 2019-11-25:
[image]

I can confirm the issue on cura version: 4.3.0 on arch linux

rburema, 2019-11-25:
@horga83 @ve7it  Is this with the AppImage version?

smartavionics, 2019-11-25:
Yes, there is still a bug in Cura that causes this. If your Cura installation directory contains both lib and lib64 sub-directories you will get the multiple suffixes. I forget now where the bug is as I just remove the lib64 directory (which is a symlink) and the problem goes away.

smartavionics, 2019-11-25:
OK, so I think the bug is caused by duplicate mime entries due to these lines in UM/Application.py...
```
        self._plugin_registry.addPluginLocation(os.path.join(self._app_install_dir, "lib", "uranium"))
        self._plugin_registry.addPluginLocation(os.path.join(self._app_install_dir, "lib64", "uranium"))
        self._plugin_registry.addPluginLocation(os.path.join(self._app_install_dir, "lib32", "uranium"))
```

When it's opening the file save dialog in LocalFileOutputDevice.py it uses a loop to add the file extensions but because there's duplicate entries in the list, you end up with duplicated extensions...
```
        for item in file_types:
            type_filter = "{0} (*.{1})".format(item["description"], item["extension"])
            filters.append(type_filter)
            mime_types.append(item["mime_type"])
            if preferred_mimetype == item["mime_type"]:
                selected_filter = type_filter
                if file_name:
                    file_name += "." + item["extension"]
```

horga83, 2019-11-25:
So really Cura should probably do a test here to see if the extension is already in the list and not add it again.

smartavionics, 2019-11-25:
Well, something has to be done but as this bug has been around for at least a year, if not 2, I don't think they care too much about it.

horga83, 2019-11-25:
so here:

if file_name:
    if item["extension"] not in file_name:
         file_name += "." + item["extension"]

ve7it, 2019-11-25:
horga83:
outstanding!   works a treat.
fix is at about line 99 in 
/usr/lib/uranium/plugins/LocalFileOutputDevice/LocalFileOutputDevice.py
remember to use spaces for the indentation to match the original file or your save button (save text on the button) wont show up.

nallath, 2019-11-26:
Yay. Fixed it (https://github.com/Ultimaker/Uranium/commit/f3a70acebb6d083980501e6abb9fafa447d9a425) . I didn't know about this issue, so that's also why I didn't fix it.

closed 2019-11-26 by nallath

smartavionics, 2019-11-28:
Did you see my comment in https://github.com/Ultimaker/Uranium/commit/f3a70acebb6d083980501e6abb9fafa447d9a425 ? I don't think that commit is good for the reason I give there.

nallath, 2019-11-28:
Yes, i just made a commit for that.

smartavionics, 2019-11-28:
Thanks for the update. I note that the filters and mime_types will still get duplicate entries but that is how it has been all along.