Ultimaker/Cura #6370: [4.3B] Support Blocker Scaling Issue

roseg0ld, 2019-09-17:
**Application version**
At least 4.2 and 4.3 beta

**Platform**
OSX Mojave

**Reproduction steps**
Insert support blocker and scale it non-uniformly along Y or Z

**Actual results**
The blue Z axis scaling handle scales along Y and vice versa 

This is not a major issue but is always inconvenient when trying to arrange support blockers on my Mac.

rburema, 2019-09-18:
Did you rotate the support blocker in relation to the main object? (Or apply any rotation/mirror beforehand?)

roseg0ld, 2019-09-18:
I did not, I rotated the main object 90 deg before adding the support blocker.

Ghostkeeper, 2019-09-19:
Yeah, I see the bug. It happens because the support blocker is set to be transforming along with the object you're placing it on (so that if you rotate that object, the support blocker rotates along). And then it is rotated back such that the initial location is always axis-aligned. If you then scale non-uniformly, the bug of #4136 shows up in the support blocker cube.

As a workaround, you can type the desired size in the text boxes in millimetres.

I'll close this as duplicate of #4136.

closed 2019-12-31 by Ghostkeeper