Ultimaker/Cura #6373: adhesion_extruder_nr as optional_extruder

maukcc, 2019-09-18:
Is there a reason that adhesion_extruder_nr is always set to T0?
Would it not be better if it would say not overridden, as all the other ones?

Ghostkeeper, 2019-09-19:
For the support extruders and the build plate adhesion extruders, there is not one single model connected to it. The "Not overridden" option lets you choose to take the extruder of the model. However since these two structures are not part of any model, it's not an option here.

Both the support and the skirt/brim/raft merge together when multiple models are involved.

maukcc, 2019-09-19:
The reason I asked is that the 1st extruder is always used for build plate adhesion, even when that extruder is not used.
So if we print a part with only the 2nd extruder, the 1st extruder is also being used for buildplate adhesion. 

Mostly it is just annoying that it is heating up 2 extruders in stead of just 1.

In case of a brim this could be a brim with a totally different material, resulting in an unusable brim as the brim will not fuse with the object.

Ghostkeeper, 2019-09-19:
Ah, then I'll close this as a duplicate of your previous ticket about this: https://github.com/Ultimaker/Cura/issues/1920

closed 2019-09-19 by Ghostkeeper