Ultimaker/Cura #6374: Filter Ironing by aspect ratio

Firedrops, 2019-09-18:
**Is your feature request related to a problem? Please describe.**
Ironing sometimes wastes time on long strips which functionally do not benefit from it.

**Describe the solution you'd like**
A filter for ironing. For example, ignore faces with aspect ratios more than 5:1, OR a minimum 'width' filter, like 1mm. Faces that would be ironed but have the shorter dimension lower than this specified width will be ignored.

**Describe alternatives you've considered**
There's currently an ironing setting for only the highest layer, but this is not fine enough for some prints.

Ghostkeeper, 2019-09-19:
Why does the aspect ratio of a shape (a shape's bounding box?) matter for whether you want to iron or not?

Firedrops, 2019-09-20:
because if it's a long thin ledge, then it's pointless to do ironing on that part.

Ghostkeeper, 2019-09-23:
Hmm. Well, I think an [opening operation](https://scikit-image.org/docs/dev/auto_examples/applications/plot_morphology.html#opening) would be more fitting then, because it won't depend on the direction of the thin ledge.

Ghostkeeper, 2021-01-18:
Perhaps it's good to realise why it's pointless to do ironing on a thin part. Because perhaps the real reason is that it's pointless to perform ironing on the walls, and the Ironing Inset is just a complete and correct solution then.

fvrmr, 2021-06-04:
We are constantly working on making Ultimaker Cura better. Our community is a big part of that by requesting new features and reporting bugs. 

To be able to focus on the most requested and needed features from our community and from Ultimaker, we have decided to remove this feature request from our backlog. We believe we will not work on this anytime soon and will therefore defer this issue.

Since Cura is open source, you or anyone else is more than welcome to work on this issue and create a pull request yourself.

Firedrops, 2023-07-09:
I think the general improvements to print quality has somewhat obsoleted the need for ironing, and the offset mentioned above should be able to catch most cases.

closed 2023-07-09 by Firedrops