Ultimaker/Cura #6375: [3.6.0] "Pause at height" resumes with a wrong speed

VijitCoder, 2019-09-18:
**Application version**
3.6.0

**Platform**
Kernel: 4.15.0-54-generic 
x86_64 bits: 64 
Desktop: Cinnamon 4.0.10
Distro: Linux Mint 19.1 Tessa

**Printer**
Creality Ender 3

**Reproduction steps**

1. Take [this model](https://www.thingiverse.com/download:3487108).

2. Add Post Processing Plugin > Pause at height. Config:

```
Pause at: layer 12
Park Print Head X: 190
Park Print Head Y: 190
Retraction: 4
Retraction Speed: 40
Extrude Amount: 0
Extrude Speed: 3.3333
Redo Layers: 1
Standby Temperature: 245
```

3. In the slicer these settings are important:
```
Infill/Wall speeds: 30/15 mm/s
Special Modes > Spiralize Outer Contour = true
Smooth Spiralized Contours = true 
```

4. Generate the gcode file.

**Actual results**

A few layers after pause will be printed on high speed, then it will be reduced to an expected value.

**Expected results**

The same printing speed as before so and after "Pause at height".

**Additional information**

The bug in the speed setting after removing pause. Here is an example of the generated code:

```
;TYPE:CUSTOM
;added code by post processing
;script: PauseAtHeight.py
;current layer: 12
M83
G1 F2400 E-4
G1 F300 Z4.2
G1 F9000 X190 Y190
G1 F300 Z15
M104 S255; standby temperature
M0;Do the actual pause
M109 S245; resume temperature
G1 F2400 E4
G1 F2400 E-4
G1 F300 Z4.2
G1 F9000 X90.765 Y95.63
G1 F2400 E4
G1 F9000         ; <----------------- Here is the wrong instruction
M82
G92 E1740.80036
;LAYER:11
;TYPE:WALL-OUTER
;MESH:cloth_vase.stl
G1 X90.765 Y95.63 E1740.80036
G1 X90.765 Y95.63 E1740.80036
G1 X91.494 Y94.841 Z2.952 E1740.88969
G1 X92.26 Y94.038 Z2.953 E1740.98196
...
```

`G1 F9000`, this command sets the breakneck speed for the hotend. And the real print speed is `G1 F900`. And it will be set, but after an indefinite number of layers further in the code. Can through 3 layer, and can - through 15. As a result, the print quality of these several layers is noticeably worse, because the other settings have not changed and the supply of plastic does not keep up with the printing speed.

Here is the [3mf.zip](https://github.com/Ultimaker/Cura/files/3627430/cloth_vase_3mf.zip)

And a photo of how the problem affects the result:

[image]

rburema, 2019-09-19:
Still happens in 4.2.1, but not with UM printers (at least not with the UM3 I tested it with), so it may be related to 'gcode-flavour'.

ninovanhooff, 2019-09-23:
@VijitCoder Thanks for your very complete bug report. We have reproduced your issue and are tracking it using ticket CURA-6799

ninovanhooff, 2019-09-23:
https://github.com/Ultimaker/Cura/pull/6400

ninovanhooff, 2019-09-24:
@VijitCoder Would you be willing to make a small testprint to validate the proposed solution? I could share a Linux AppImage testbuild with you. (expected tomorrow)

VijitCoder, 2019-09-24:
I can just check the gcode.

Printing in the same circumstances a bit awkward (change the nozzle, recreate plugin settings). And I'm not sure that it is really necessary.

LipuFei, 2019-09-25:
Fixed by #6400

closed 2019-09-25 by LipuFei

ninovanhooff, 2019-09-27:
@VijitCoder Since we do not have your printer available we have used the same method to verify the fix. Since it has been merged in master you can expect the fix to be included in the 4.4 release.

VijitCoder, 2019-09-27:
> we have used the same method of inspecting gcode to verify the fix

That should be enough, I think.

> you can expect the fix to be included in the 4.4 release.

No updates in plugin for Cura 3.x? 

Cura 4.x has a terrible interface. I wouldn't want to switch to it just for the sake of a fixed plugin, if it possible. Just as a humble wish :)

Ghostkeeper, 2019-09-30:
We're not releasing new versions of Cura 3.x unless maybe there is a critical security issue. But there'd have to be a really good reason for our users to not update their Cura versions even then.

That said, the API of the post-processing scripts hasn't changed much (any?) since 3.6. There's a decent chance that you can just take our [most recent version of the script](https://github.com/Ultimaker/Cura/blob/master/plugins/PostProcessingPlugin/scripts/PauseAtHeight.py) and paste that into your Cura installation. However since you're on Linux you'd need to change that into the mounted AppImage which requires some trickery. I'm not sure on how to do that.

VijitCoder, 2019-09-30:
AppImage is not a problem.

And yeap, It works :) Thanks.