Ultimaker/Cura #6376: Travel avoid

jokamikaze, 2019-09-18:
Prioritize (or put an select box) the "avoid printed parts" over the "avoid supports", if both are enable Cura prefer avoid supports.

[image]

Ghostkeeper, 2019-09-19:
It's currently avoiding all of them at the same time. This cuts the computation time for calculating combing down a lot.

adamsiem, 2020-04-02:
@Ghostkeeper Is this expected behavior in my screenshot for Travel Avoid? I have Combing Mode: All, but it does not comb for the few supports created in the slot at the nearest end of the fins of the model.
[image]

Ghostkeeper, 2020-04-06:
I'll refer to my answer here then: https://github.com/Ultimaker/Cura/issues/2401#issuecomment-609764762

Your comment is off-topic for this issue thread. This issue thread is about prioritising which parts of the print it hits if collisions are inevitable and combing is impossible.

This would mean that CuraEngine would first need to calculate if there is a valid path without hitting any walls or support, and if there is no such valid path it should calculate again to see if there is a valid path that doesn't hit any walls but may hit some support. We don't think this case occurs very often and it would mean a significant slowdown for slicing for everyone, so we've chosen to defer it until we have a better system to compute combing paths with priorities.

closed 2023-06-13 by github-actions[bot]