Ultimaker/Cura #6377: Weird holes/artifacts on surface after slicing for monoprice select mini v2.

Gzalo2188, 2019-09-18:
**Application version**
4.2.1

**Platform**
[Personal Laptop]
Windows 10
GTX 1650

[School Desktops]
Windows 10
GTX 1080

**Printer**
Monoprice Select Mini v2

**Reproduction steps**
Happens with most meshes that have a shell.

**Actual results**
Defects appear on the surface of the print. Usually as big holes or smaller dips.

**Expected results**
Surface should be smooth.

**Additional information**
I run a maker space at a community college so we have multiple computers with cura. The issue seems to appear frequently on all computers when slicing for the monoprice select mini v2. However when slicing for the ultimaker 2+ the issue doesnt show up.

[Cura_Issue.zip](https://github.com/Ultimaker/Cura/files/3627884/Cura_Issue.zip)
Ive attached both the test object and a project file. Also here's a picture of the [issue](https://i.imgur.com/wGfu7Uy.png).

smartavionics, 2019-09-18:
Hello @Gzalo2188 , in that profile you are using coasting, if you turn that off the large gaps shown on the layer view go away. Personally, I never use coasting but if you wish to use it, perhaps you should reduce the amount of coasting?

With coasting...

[image]

Without coasting...

[image]

Gzalo2188, 2019-09-18:
@smartavionics Thanks! That definitely solved the issue.

For some reason coasting is turned on by default with the select mini profile when switching to expert mode.

closed 2019-09-19 by rburema