Ultimaker/Cura #6378: [4.2.1] Per Model Setting assigned to wrong model(s) after slicing.

Ugars, 2019-09-18:
**Application version**

Cura Version 4.2.1

**Platform**

Windows10 64-bit

**Printer**

Using Creality Ender-3

**Reproduction steps**

I used per model settings for my 4 models on the virtual build plate, for A+B+C models I used per model setting to change the print speed from 60 mm/s to 30 mm/s, for model D I used per model setting to change the infill ratio from 10% to 20% and sliced and transferred the file as usual.

**Actual results**

A+B+C models which are supposed to slowed down are printed at normal speed, but model D printed at slower speed, as you can guess this doesn't expected to happen like this.

**Expected results**

[Cura_4.2.1 Per Model Setting Problem.zip](https://github.com/Ultimaker/Cura/files/3628490/Cura_4.2.1.Per.Model.Setting.Problem.zip)

**Additional information**

rburema, 2019-09-19:
Could you please attach the .3mf (saved, not exported)? The problem is likely to be _before_ gcode generation begins.

Ghostkeeper, 2019-09-19:
I'm not able to reproduce this. I loaded 4 models, changed 3 of them to 30mm/s and one of them to 60% infill rate (it was already 20% to start with for me).

Shown in layer view using the feedrate colour scheme, the three objects were made to print slower and the last object has a greater infill density.

Ugars, 2019-09-20:
> I'm not able to reproduce this. I loaded 4 models, changed 3 of them to 30mm/s and one of them to 60% infill rate (it was already 20% to start with for me).
>
> Shown in layer view using the feedrate [...]

Everything seemed fine in Cura but in actual printing it was not. I don't know may be it is because of the new marlin (1.1.9) update I made a while ago before this print. Sorry if I take your time with my mistake...

Ghostkeeper, 2019-09-23:
Maybe it's interpreting the g-code differently? Maybe it's not properly handling the limits?

You could try enabling or disabling the Acceleration Control and Jerk Control settings, see if that changes anything. Those tend to be somewhat more complex to handle in Marlin.

closed 2020-06-19 by no-response[bot]