Ultimaker/Cura #6379: Support Problem

abrooks-Accutime, 2019-09-19:
Cura does not add support where it needs support. This problem was also found in Cura 4.2.0 and Cura 4.2.1
[image]
[image]

nallath, 2019-09-19:
Could you also share the model and the settings that you used to get this?

abrooks-Accutime, 2019-09-19:
Is there an email I can send this to since I don't want this model in the public?

nallath, 2019-09-19:
Sure, you can send it to j.vankessel@ultimaker.com

nallath, 2019-09-20:
I don't know what the problem is, but I've tried to slice it with 4.2 and 4.3 (although I am running it from source) with the profile that you provided. In both cases I got correct support:

4.2:
[image]

4.3
[image]

I'm guessing you must have other changes than the ones you sent me. Could you share a project file of the setup where the issue occurs?

abrooks-Accutime, 2019-09-20:
Let me know if you can see the problem know. I don’t understand what going on here and how you can see the normal support.

Best Regards,

Andrew Brooks
Electrical Engineer

Accu-Time Systems, Inc.
175 West Road, Suite #4
Meadowview Plaza
Ellington, CT 06029
T. (860) 870-5000
D. (860)375-2028
Web: www.accu-time.com<http://www.accu-time.com/>
[ATS_logo_signature]

[Cura42Support]<https://user-images.githubusercontent.com/3922611/65308649-d1013d80-db8a-11e9-99fc-f24f8993ef8d.png>

[SupportCura43]<https://user-images.githubusercontent.com/3922611/65308754-060d9000-db8b-11e9-8dac-f69364ce9fd0.png>

Ghostkeeper, 2019-09-23:
We were able to reproduce this issue last Friday. It turns out to be a bug related to stair stepping. We'll raise this and see if a fix can be made, but it won't make it for 4.3.

As a workaround, you can set the Support Stair Step Height setting to 0.

cnfengjie, 2020-02-28:
> We were able to reproduce this issue last Friday. It turns out to be a bug related to stair stepping. We'll raise this and see if a fix can be made, but it won't make it for 4.3.
> 
> As a workaround, you can set the Support Stair Step Height setting to 0.

Sorry anybody can help to explain if change “Support Stair Step Height‘’  to ‘0’, What will be impact? Can I rest assured change it?

abrooks-Accutime, 2020-02-28:
This issue has been fixed on Cura 4.5 Beta

cnfengjie, 2020-02-29:
> This issue has been fixed on Cura 4.5 Beta

Wow, good news, could you give me a download link？ I always don't know where I can get version update news or find new version~

mahtDFR, 2020-03-02:
Internal discussion on this here: CURA-6989

abrooks-Accutime, 2020-03-02:
cnfengjie,

I think I'm getting you confused to what my issue was at hand here. I had an issue where there will be no support from the build plate to the 90 degree overhang. All I know is if I have any issues with the new software, I would just go back to Cura version 4.1

Ghostkeeper, 2020-03-03:
I don't think this is fixed in Cura 4.5 yet. Our ticket on this is still open at least.

nallath, 2020-06-22:
Our internal ticket for this has finally been resolved. This should be fixed for 4.7

closed 2020-06-22 by nallath