Ultimaker/Cura #6381: Special Modes: Mold has inaccurate Dimensions

nsmela, 2019-09-19:
**Application version**

Cura 4.0.0

**Platform**

Windows 7

**Printer**

Fusion3 F410
using 0.60 mm nozzle

**Reproduction steps**

Created a 100 x 5 x 100 mm cube
used the Special Modes: Mold setting
when print was done, filled with 2-part silicone and cured
once cured, removed the plastic mold to have a silicone block

**Actual results**

measured weight (43 grams, expected 53 grams)
measured width (~4.0 mm, expected 5 mm)

**Expected results**

measured weight (43 grams, expected 53 grams)
measured width (~4.0 mm, expected 5 mm)

**Additional information**

Suspect that when molding, the printer has the nozzle head move along the edge of the mold instead of offset to account for the width of the nozzle.

rburema, 2019-09-20:
What's your 'Minimal Mold Width' setting say? I think that's supposed to compensate for the effect you mention.

nsmela, 2019-09-20:
[cube_mold_test.zip](https://github.com/Ultimaker/Cura/files/3636444/cube_mold_test.zip)

Does that file work?

The hint for _Minimal Mold Width_ says it's the minimum distance between the outside of the mold to the outside of the model.  Am I missing something?

Ghostkeeper, 2019-09-23:
Minimal Mold Width is essentially just the thickness of the shell. It just says "minimal" because it can be made thicker to allow for the mould angle.

In your .zip file the block has a width of 10mm. When I'm slicing it, it's generating a cavity with the correct width. Here is a snippet of g-code that demonstrates the width (line 125868 and onwards):

```
G1 F1800 X236.14 Y180.2 E8.94751
G1 X236.14 Y169.8 E0.76099
G1 X113.86 Y169.8 E8.94751
G1 X113.86 Y180.2 E0.76099
```

So the centreline ranges from Y=180.2 to Y=169.8. That's a difference of 10.4mm. Your line width is set to 0.4mm. Two times half the line width will fall between the centrelines, so `2 * 0.4/2 = 0.4`. So `10 + 0.4 = 10.4`, thus the distance seems to be correct.

I think you're simply experiencing a build-up of inaccuracies between the printer, the moulding material and the cast material.

Ghostkeeper, 2019-09-23:
We're unable to find any inaccuracies in the g-code so far. Can you point out where the g-code is inaccurate? Or maybe you know of a better way to print this without the real-world consequences of inaccuracies/shrinkage?

nsmela, 2019-09-23:
Nope, I think you isolated the issue to being our printer rather than the gcode. Thank you for your help and sorry for the confusion!

diegopradogesto, 2019-09-24:
Thanks for the report @nsmela . I'm closing the issue since it seems that your question was answered. Feel free to reopen if there are more related questions.

closed 2019-09-24 by diegopradogesto