Ultimaker/Cura #6383: Tool/active nozzle is defined before the custom start-Gcode

joaomamede, 2019-09-20:
I have a two nozzle two extruder system.

This is my starting g-code:
`; Ender 3 Custom Start G-code
M104 T0 S{material_print_temperature_layer_0, 0} ; Set Extruder temperature
M104 T1 S{material_print_temperature_layer_0, 1} ; Set Extruder temperature
M140 S{material_bed_temperature_layer_0} ; Set Heat Bed temperature
M190 S{material_bed_temperature_layer_0} ; Wait for Heat Bed temperature
M109 T0 S{material_print_temperature_layer_0, 0 } ; Wait for Extruder temperature
M109 T1 S{material_print_temperature_layer_0, 1} ; Wait for Extruder temperature
;G90 ; octolapse
G28 ; Home all axes
G29 L1 ; UBL loadd
G29 J; Auto lvl ubl 3points
;G29
T0
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5.1 Y40 Z0.2 F5000.0 ; Move to start position
G1 X5.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X5.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
G1 X5.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 F200 E-2
T1
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10.1 Y40 Z0.2 F5000.0 ; Move to start position
G1 X10.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X10.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
G1 X15.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 F200 E-2`

Basically Tool is set by cura before the starting G-code meaning that after my priming custom code it always makes the first layer with T1 (as I left it from the purge commands).

1- Is it possible to have the tool selection (T0 or T1) from cura slicing after the starting G-code
2- As a workaround, is there a variable like {active_tool} that I can put at the end of my custom start g-code?

This is how it looks like after slicing
`;FLAVOR:Marlin
;TIME:4962
;Filament used: 0m, 3.14559m
;Layer height: 0.12
;MINX:88.2
;MINY:109.7
;MINZ:0.2
;MAXX:146.8
;MAXY:138.2
;MAXZ:40.28
;Generated with Cura_SteamEngine 4.2.1
T1 ; This is the problem right here!!!!
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M104 T0 S200 ; Set Extruder temperature
M104 T1 S210 ; Set Extruder temperature
M140 S45 ; Set Heat Bed temperature
M190 S45 ; Wait for Heat Bed temperature
M109 T0 S200 ; Wait for Extruder temperature
M109 T1 S210 ; Wait for Extruder temperature
;G90 ; octolapse
G28 ; Home all axes
G29 L1 ; UBL loadd
G29 J; Auto lvl ubl 3points
;G29
T0
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X5.1 Y40 Z0.2 F5000.0 ; Move to start position
G1 X5.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X5.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
;G1 X5.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 F200 E-2
T1
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10.1 Y40 Z0.2 F5000.0 ; Move to start position
G1 X10.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X10.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
G1 X15.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 F200 E-2
; End of my custom Code!
M141 S28
G92 E0
G92 E0 ; G92 E0 repeats here I don't know why
; The T0 or T1 from slicing should be around here instead of the beggining
G1 F2700 E-4
;LAYER_COUNT:335
;LAYER:0
M107
G0 F3600 X91.367 Y111.252 Z0.2
;TYPE:SKIRT
G1 F2700 E0
[...] 
`

rburema, 2019-09-20:
For 2., you can put `T{initial_extruder_nr}` in your start code.

rburema, 2019-09-20:
Otherwise, this is a duplicate of https://github.com/Ultimaker/Cura/issues/4526, which is a won't fix (see that issue for an explanation.)

joaomamede, 2019-09-25:
T{initial_extruder_nr} is not always working.

I added it at the end of my "start g-code",
Still there is a mismatch between the Tool selected in the beginning (which is correct) and the one defined by the variable.

Example:

>;FLAVOR:Marlin
> ;TIME:28652
> ;Filament used: 5.32274m, 17.1985m
> ;Layer height: 0.16
> ;MINX:27.207
> ;MINY:31.117
> ;MINZ:0.2
> ;MAXX:206.326
> ;MAXY:203.889
> ;MAXZ:47.12
> ;Generated with Cura_SteamEngine 4.2.1
> **T0 ; <- Added at beggining and it's correct**
> M82 ;absolute extrusion mode
> ; Ender 3 Custom Start G-code
> M104 T0 S197 ; Set Extruder temperature
> M104 T1 S197 ; Set Extruder temperature
> M140 S55 ; Set Heat Bed temperature
> M190 S55 ; Wait for Heat Bed temperature
> M109 T0 S197 ; Wait for Extruder temperature
> M109 T1 S197 ; Wait for Extruder temperature
> ;G90 ; octolapse
> G28 ; Home all axes
> G29 L1 ; UBL loadd
> G29 J; Auto lvl ubl 3points
> ;G29
> T0
> G92 E0 ; Reset Extruder
> G1 Z10.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X10.1 Y40 Z0.2 F5000.0 ; Move to start position
> G1 X10.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
> G1 X10.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
> G1 X10.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
> G92 E0 ; Reset Extruder
> G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 F200 E-3
> T1
> G92 E0 ; Reset Extruder
> G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 X13.1 Y40 Z0.2 F5000.0 ; Move to start position
> G1 X13.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
> G1 X13.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
> G1 X13.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
> G92 E0 ; Reset Extruder
> G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
> G1 F200 E-3
> **T1 ; <- Added by T{initial_extruder_nr}**
> M141 S28
> G92 E0
> G92 E0
> G1 F3300 E-5
> ;LAYER_COUNT:293

Ghostkeeper, 2019-09-25:
The `initial_extruder_nr` gets set to the extruder with the lowest index that is used by your current print. So if there is a `T0` in the g-code but `initial_extruder_nr` becomes `1`, then we have a bug. Are the disallowed (shadow) areas correct on your build plate? If you're only using the second extruder, it should not show the areas where the first extruder is unable to reach.

joaomamede, 2019-09-27:
So. If I have.say. Brimming in T1, And up top T1 as it should. Sometimes I have a mismatch the other way around. I can replicate this quite a few times. I just can't pinpoint why, this is happening. 
Is there a varible T{get_current_tool_nr} kind ot thing?

Another example where I start printing with T1 and then I have a T0 with initial_extruder_nr (reversed from previous report)

```
;FLAVOR:Marlin
;TIME:42858
;Filament used: 12.3862m, 25.2655m
;Layer height: 0.2
;MINX:33.8
;MINY:41.783
;MINZ:0.2
;MAXX:205.274
;MAXY:177.692
;MAXZ:83.2
;Generated with Cura_SteamEngine 4.2.1
T1
M82 ;absolute extrusion mode
; Ender 3 Custom Start G-code
M104 T0 S200 ; Set Extruder temperature
M104 T1 S200 ; Set Extruder temperature
M140 S55 ; Set Heat Bed temperature
M190 S55 ; Wait for Heat Bed temperature
M109 T0 S200 ; Wait for Extruder temperature
M109 T1 S200 ; Wait for Extruder temperature
;G90 ; octolapse
G28 ; Home all axes
G29 L1 ; UBL loadd
G29 J; Auto lvl ubl 3points
;G29
T0
G92 E0 ; Reset Extruder
G1 Z10.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X10.1 Y40 Z0.2 F5000.0 ; Move to start position
G1 X10.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X10.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
G1 X10.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 F200 E-3
T1
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 X13.1 Y40 Z0.2 F5000.0 ; Move to start position
G1 X13.1 Y220.0 Z0.2 F1500.0 E15 ; Draw the first line
G1 X13.4 Y220.0 Z0.2 F5000.0 ; Move to side a little
G1 X13.4 Y40 Z0.2 F1500.0 E30 ; Draw the second line
G92 E0 ; Reset Extruder
G1 Z5.0 F3000 ; Move Z Axis up little to prevent scratching of Heat Bed
G1 F200 E-3
T0
M141 S27
G92 E0
G92 E0
```

joaomamede, 2019-09-27:
About the shadow areas. I have the offset set by firmware in my case. But when I used cura to do it, I have the impression that it was flipped.

 (after testing) - It I put the offset to 18mm I have shading in both sides not only one.

Ghostkeeper, 2019-09-27:
> Is there a variable T{get_current_tool_nr} kind ot thing?

No, there isn't. Cura assumes that the printer starts on `T0`, but in reality it knows just as little about the starting state of the printer as yourself. So instead of `{current_extruder_nr}` you would just fill in whatever you typed last in your start g-code manually.

joaomamede, 2019-09-27:
But cura first line actually selects the right tool all the time. Sometimes T0 sometimes T1 (see two examples above).

Ghostkeeper, 2019-09-30:
Maybe it's best if you can share a project file then so that we can look for ourselves to see what's going wrong. Cura should start printing with the first extruder (lowest #) that is used in the print.

CumpyRules, 2020-01-17:
Hi,

> 2- As a workaround, is there a variable like {active_tool} that I can put at the end of my custom start g-code?

Usually, the first print ist the Skirt or Brim. You can set in the GCode Settings for "Startcode" for "T" instead the {initial_extruder_nr} Keyword, the {adhesion_extruder_nr}.

No not the active Extruder will get set, but the Extruder which will be used for the Skirt.
I mean that should work for you !

Greetz CR

Ghostkeeper, 2020-01-17:
For the upcoming version we've fixed a bug with `initial_extruder_nr` being wrong if the adhesion extruder was changed.

eugr, 2020-02-10:
@Ghostkeeper - do you have a reference to the commit? I'm trying dual extrusion with my Creator Pro and whenever I activate both extruders, initial_extruder_nr is always 0, even if I change the skirt/brim to print with 1 and that messes up all startup scripts.

mahtDFR, 2020-02-10:
@eugr https://github.com/Ultimaker/Cura/commit/d18c0703b49aba38da18810c1df226166f61cf63#diff-a898e447d66fcc2aadbb675de2107611

eugr, 2020-02-10:
@mahtDFR - thanks, that should fix it! Now if we could have some
way to detect if both extruders are enabled... So far I had to create a separate profile for dual extrusion only, so I could do initial prime and wipe for both nozzles in start GCode...

Ghostkeeper, 2020-02-12:
In an inheritance function (if you're writing the profile with a text editor) you can ask for `extruderValues("extruder_nr")` to see which extruders are active and then adjust the setting value based on that. That's not possible from the GUI though.

eugr, 2020-02-12:
@Ghostkeeper - wow, I somehow missed it. Yeah, I'm trying to create a "proper" profile that I hope I could commit to Cura main repository at some point in the future. I'll see what I can do with that function. My profile is over here for now: https://github.com/eugr/Flashforge-for-Cura

Ghostkeeper, 2020-02-14:
It's no wonder you missed it. These things aren't documented all too well.

eugr, 2020-02-14:
@Ghostkeeper - it worked! I have a fully-functioning profile now... I wonder what's the process to make it a part of Cura distribution? Should I fork the repo and submit a pull request? I still need to do some more extensive testing, but I've been using my profile exclusively for about a month now and it works much better than other slicers that support FFCP natively. So once I'm happy with dual extrusion I think it would benefit the users if I contribute it to Cura itself...

fieldOfView, 2020-02-15:
> Should I fork the repo and submit a pull request?

This is The Way.

Ghostkeeper, 2020-02-18:
This is The Way.

eugr, 2020-02-26:
@Ghostkeeper  - Is there any way to specify profile version? There is "version" field, but it's set to "2" in every profile I've seen, so I guess that's the file format version, not the actual profile version?

Ghostkeeper, 2020-02-28:
There is `version` which is indeed the file format version. There is also `setting_version` which is the version number for the list of settings. We need to increment the `setting_version` with almost every release; at least if there is any deleted or renamed setting. Since that would require us to also delete or rename these settings in the user's config directory.

For pull requests submitted to Cura we require you to use the most recent `setting_version`. The ones that are merged in are being supported by the developers, so we'll change the `setting_version` when needed and then also delete/rename any settings if necessary.

There is no version number for the actual profiles in Cura. We don't track how often a profile was modified. The only exception is the XML material profiles, since those need to be sent to the Ultimaker printers as well so they need to know when Cura needs to send a newer profile to store in the printer.

closed 2024-12-07 by github-actions[bot]