Ultimaker/Cura #6384: Failed to probe OpenGL

gangemi, 2019-09-20:
**Application version**
4.2.1

**Platform**
Windows 10

**Printer**
Ender 3 (though not important here)

**Reproduction steps**
Open Cura

**Actual results**
Error box: Failed to probe OpenGL.  Could not probe OpenGL. This program requires OpenGL 2. or higher. Please check your video card drivers.

**Expected results**
Open Cura.

**Additional information**
Similar to issue #5474, except I'm just running a regular desktop PC.  I've run 4.2.1 before, but today I installed a USB dock with display ports (specifically the Plugable UD-3900).  Plugable says they sometimes have problems with this as OpenGL doesn't respond with a version (or something).  They feel they've solved the problem with a recent Windows update, but obviously it isn't working for Cura.  Ticket here (https://support.displaylink.com/forums/287786-displaylink-feature-suggestions/suggestions/15869604-fix-some-applications-not-running-on-laptop-gpu?page=2&per_page=20).  I've installed KB4512941 and this isn't fixed.

I'll continue to work with them, but was hoping there's something you could do on your end.  I'm sure I'm not the only one working off a USB dock.  (for reference, USB dock is just so I can switch my monitors, keyboard, and mouse to a laptop when I need to bring it home to work - I could hook my monitors back up to my desktop's HDMI, but that means getting down behind my desk and pulling wires every time I want to use Cura)

rburema, 2019-09-20:
> but today I installed a USB dock with display ports (specifically the Plugable UD-3900)

What we sometimes see with the USB docks (different type though) we're using in the office (but only sometimes for some reason), is that the 'main screen' needs to be the laptop screen in order for programs to run with the gpu-card (not the internal intel one).

fieldOfView, 2019-09-20:
From the Plugable site:

Microsoft and DisplayLink (the maker of the chip in our dock and author of the driver) have identified a bug within the Windows 10 Anniversary and Creators Update. With the Anniversary or Creators Update to Windows 10, Microsoft has now integrated support for the DisplayLink technology used in our docking station into Windows itself, and in certain cases applications which make use of OpenGL may not work as expected.For example, the SketchUp application requires the use of graphics adapter that supports a standard known as OpenGL -> http://help.sketchup.com/en/article/114278

In some instances, Windows mistakenly directs the tasks intended for the system’s internal graphics adapter (which does support OpenGL) to the DisplayLink-based docking station. Our docking station is not a traditional graphics adapter in and of itself (it relies on the CPU and internal graphics adapter to work), and this results in the incorrect behavior or error message. DisplayLink speaks to this issue directly here -> http://support.displaylink.com/knowledgebase/articles/942862-my-device-is-not-working-properly-on-windows-10-an (under the section ‘OPENGL 1.1 REPORTED INSTEAD OF 4.X’)

Pending Microsoft fixing the issue within Windows itself, there are two possible workarounds. The first is to configure Windows to have the ‘Main’ display set to a display directly connected to your internal graphics adapter, whether that is an internal laptop display or another external display connected to one of the systems built-in video outputs (a quick guide to doing so is here -> https://youtu.be/7nnKztRZXsw).If that doesn’t prove a suitable workaround, booting the system without the dock connected, launching SketchUp (or any other OpenGL application), and then connecting the dock may also help.Please note that the HDMI port on this dock is unaffected by this as that output is driven directly from the host computer graphics chipset over USB-C Alternate Mode video.

gangemi, 2019-09-20:
Thanks both. But these fixes assume I have a laptop. In effect a third monitor not connected to the usb. I’m trying to use Cura on my desktop, that doesn’t have its own built-in monitor.

fieldOfView, 2019-09-20:
What GPU does your desktop have?

gangemi, 2019-09-21:
It's an i7-2600 Intel.  I believe the GPU is built into the chip.

fieldOfView, 2019-09-22:
Having looked into it more, I don't think there is anything Cura can do from its end. The OS tells Cura the GPU only supports OpenGL 1.1 (even though the GPU supports OpenGL 3.1, but the USB dock gets in the way). Cura needs OpenGL 2.1 or better.

Pluggable, DisplayLink or Microsoft will have to solve this for you (or you will have to accept that you will need something else to switch your displays).

Ghostkeeper, 2019-09-23:
Hey, you have the same CPU as me! But you don't want to use the embedded GPU in that. Its OpenGL implementation is much slower than software rendering! Oh well.

I think we'll close this issue since the issue seems to be in your operating system rather than with Cura.

closed 2019-09-23 by Ghostkeeper