Ultimaker/Cura #6386: [4.2.1] Brim creation issue

Dougal57, 2019-09-20:
**Application version**

**Platform**

**Printer**

**Reproduction steps**

**Actual results**

**Expected results**

**Additional information**

Ghostkeeper, 2019-09-24:
Most likely this 1.46466mm is larger than half of the initial layer height. Is your initial layer height 0.27mm? Then it'll make the slice at 0.27mm/2 = 0.135mm. Only if you set the Slicing Tolerance to Inclusive will it take the first slice at 0mm.

I think the support was omitted because of the Support Z Distance.

Dougal57, 2019-09-24:
Hi Ghostkeeper

I note this says "

Thank you for taking the trouble to understand what must have been 
minimal information. This was my first attempt to report an issue, and I 
made a simple 'newbie' mistake. I attempted to follow the guidelines, 
but (for reasons obvious to me afterwards <!>) none of the lengthy 
explanation I wrote actually appeared in the final result. In deep 
embarrassment, I immediately closed the state.

I think you are right about the Support Z Distance.

My Malyan 200 is quite hard-worked, and has a PEI layer on the heated 
bed.  On this occasion I was using a layer height of 0.0875mm, an 
initial layer height of 0.2625mm as you guess, and a support z-distance 
of 0.1mm. When models are as delicate as this one is, I find I have to 
use a brim, sometimes a substantial brim, to support the printing.

I use OpenScad for my designs, and on this occasion had rotated the 
final (somewhat delicate) model by 92 degrees to approximately align 
four of its 'feet' with the printing base plate. This was in a hasty 
attempt to finish a quick trial print before bedtime. This is something 
I have 'got away with' on other designs many times before: either a brim 
will be generated, or support and a brim will be generated: either is 
fine for a quick trial.

I later calculated that my rotation should have been 91.7457 degrees: 
quite by chance two of the 'feet' were therefore 0.146466mm away from 
the print bed.  And 0.146466 is indeed larger than half the initial 
layer height.

I classified this as a 'bug' because it seems to me that, if a brim is 
needed to support something resting on the print bed, and a brim is 
needed to support any support, a brim is surely also needed on those 
very rare occasions (like this one) when the print bed itself is 
'considered to' provide the support?

By the way, I note now that 0.146466 is also larger than the Support Z 
distance, so shouldn't support have been generated in any case?

regards

Ghostkeeper, 2019-09-24:
> Thank you for taking the trouble to understand what must have been
minimal information. This was my first attempt to report an issue, and I
made a simple 'newbie' mistake. I attempted to follow the guidelines,
but (for reasons obvious to me afterwards <!>) none of the lengthy
explanation I wrote actually appeared in the final result. In deep
embarrassment, I immediately closed the state.

Yeah we have a template that has HTML comments in them, and you put your explanations inside the comments. I removed the comment markers so your explanations became visible. Happens more often.

> I use OpenScad for my designs, and on this occasion had rotated the
final (somewhat delicate) model by 92 degrees to approximately align
four of its 'feet' with the printing base plate. This was in a hasty
attempt to finish a quick trial print before bedtime. This is something
I have 'got away with' on other designs many times before: either a brim
will be generated, or support and a brim will be generated: either is
fine for a quick trial.

I often find it easier to move the model down slightly underneath the build plate. All layers underneath the build plate are left out so it'll stick properly. Of course the total height of the print will be inaccurate then though.

> By the way, I note now that 0.146466 is also larger than the Support Z
distance, so shouldn't support have been generated in any case?

Sadly, the support is still limited to be printed on a layer so it can't be printed halfway along the layer's thickness. So we're rounding it. I believe we're currently calculating the Z distance by dividing it by the layer height and then looking that number of layers higher if something needs to be supported. This would be incorrect for the initial layer. But it'll be difficult to make that work right because not the entire model is near the initial layer. We can look into that.

closed 2024-12-07 by github-actions[bot]